carla-simulator/carla #1227: Carla 0.9.3: Segmentation fault (core dumped)

krdindorkar19, 2019-02-08:
[image]

**Command used to run carla server:**
```./CarlaUE4.sh /Game/Carla/Maps/Town01 -carla-server --carla-port=2000 -benchmark -fps=10 -windowed -ResX=640 -ResY=480````

**also facing same issue while using Cralasim/carla-0.9.3 docker container.**

```
[2019.02.08-09.55.08:330][  0]LogRenderer: Warning: Gaussian DOF algorithm is deprecated and will be removed from deferred shading renderer. Consider using the Circle DOF method in post process settings.
[2019.02.08-09.55.11:167][  0]LogLinux: Setting swap interval to 'Immediate'
[2019.02.08-09.55.11:167][  0]LogSlate: Slate font cache was flushed
[2019.02.08-09.55.12:336][  7]LogCarla: Loading OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/Town01.xodr'
[2019.02.08-09.55.12:492][  8]LogCarla: Spawning actor 'vehicle.carlamotors.carlacola'
[2019.02.08-09.55.13:240][ 10]LogCarla: Spawning actor 'sensor.other.collision'
[2019.02.08-09.55.13:269][ 11]LogCarla: Loading OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/Town01.xodr'
[2019.02.08-09.55.13:476][ 13]LogCarla: Spawning actor 'sensor.other.gnss'
[2019.02.08-09.55.13:644][ 14]LogCarla: Loading OpenDrive file '../../../CarlaUE4/Content/Carla/Maps/OpenDrive/Town01.xodr'
[2019.02.08-09.55.13:843][ 16]LogCarla: Spawning actor 'sensor.camera.rgb'
[2019.02.08-09.55.13:843][ 16]g.TimeoutForBlockOnRenderFence = "300000"
[2019.02.08-09.55.13:960][ 18]LogRenderer: Reallocating scene render targets to support 1280x720 Format 10 NumSamples 1 (Frame:18).
[2019.02.08-09.55.20:289][ 25]LogRHI: Error: Failed to link program. Current total programs: 273 binary bytes: 758517497
  log:
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/home/kpit/Carla0.9.3/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718] 
Failed to link program. Current total programs: 273, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=98832 LargeMemoryPoolOffset=298096 
Malloc Size=66130 LargeMemoryPoolOffset=364242 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Unlingius, 2019-02-08:
Have you compiled it yourself? Did you changed any rendering settings in Unreal Engine editor before (in another version carla, for example)? If yes, maybe it will help to remove ~/.config/Epic/Unreal and all similar dirs with 'unreal'

Dhairyashil-kpit, 2019-02-13:
Hi. facing same issue in carla:0.9.3 docker container.

```
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/home/kpit/Carla0.9.3/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718]
```

krdindorkar19, 2019-02-13:
Hi @Unlingius,
Sometimes server runs and when I run manual_control.py, it works. However when I try with my code, it crashes again but same code works fine with Carla 0.9.2

I have changed get_vehicle_control() to get_control()

I have attached my file, Kindly let me know what other changes I need to make
[client_world_9.3.txt](https://github.com/carla-simulator/carla/files/2859537/client_world_9.3.txt)

SAmmarAbbas, 2019-02-18:
Getting the same error. I am running with Core i7 8550-U and Intel UHD 620 on Ubuntu 16.04. I compiled it myself and did not change any settings.

Ebrahim1604, 2019-03-20:
Any updates on this issue? Facing similar issues when downloaded the assets from the getting started page of Carla AD challenge.
https://carlachallenge.org/get-started/

SAmmarAbbas, 2019-03-20:
@Ebrahim1604 I think I was getting the error at the point the op has mentioned; not while downloading the assets. Also, I think the error was because I didn't have a dedicated gpu. I got it to run on my other laptop which has GTX 1050 Ti.

Ebrahim1604, 2019-03-21:
@SAmmarAbbas yes i also got the error exactly at the point op mentioned, I mentioned about downloading the assets to state that i have not built it from source. Anyway, i have an Amd graphic card, so maybe that's the issue. Need more people to confirm it.

Johnnyswong, 2019-04-02:
The Carla docker container requires nvidia-docker to call the OpenGL library. Install the nvidia-docker will fix this issue.

15327311512, 2019-04-07:
I had downloaded  the nvidia-docker,but i still face this problem

Johnnyswong, 2019-04-10:
@nsubiron  Can you focus on this issue? It seems unsolved.

lebu, 2019-04-23:
I'm also facing this issue in 0.9.5, while 0.9.2 works.

nsubiron, 2019-04-29:
Closing this one, let's continue in #1210.

closed 2019-04-29 by nsubiron