carla-simulator/carla #1230: Fail to run the Windows version 0.8.2 and 0.8.4

linbaiwpi, 2019-02-09:
Hello
I download the Windows Compiled version 0.8.2. But got the issue "ERROR: (localhost:2000) failed to connect: [WinError 10061] No connection could be made because the target machine actively refused it". On 0.8.2, I could solve this by adding a Example.CarlaSettings.ini.

Then I encountered the following error:

(carla) C:\Users\lbai\Downloads\CARLA_0.8.2>python PythonClient\client_example.py
INFO: listening to server localhost:2000
CarlaClient connected
Starting new episode...
Traceback (most recent call last):
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 174, in _read_sensor_data
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "PythonClient\client_example.py", line 248, in <module>
    main()
  File "PythonClient\client_example.py", line 235, in main
    run_carla_client(args)
  File "PythonClient\client_example.py", line 113, in run_carla_client
    measurements, sensor_data = client.read_data()
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 127, in read_data
    return pb_message, dict(x for x in self._read_sensor_data())
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 127, in <genexpr>
    return pb_message, dict(x for x in self._read_sensor_data())
RuntimeError: generator raised StopIteration

(carla) C:\Users\lbai\Downloads\CARLA_0.8.2>python PythonClient\driving_benchmark_example.py
 WARNING: running the basic driving benchmark, to run for CoRL 2017 experiment suites, you should run python driving_benchmark_example.py --corl-2017
Traceback (most recent call last):
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 174, in _read_sensor_data
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "PythonClient\driving_benchmark_example.py", line 92, in <module>
    args.host, args.port)
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\driving_benchmark\driving_benchmark.py", line 294, in run_driving_benchmark
    benchmark_summary = benchmark.benchmark_agent(experiment_suite, agent, client)
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\driving_benchmark\driving_benchmark.py", line 129, in benchmark_agent
    + '.' + str(end_index))
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\driving_benchmark\driving_benchmark.py", line 206, in _run_navigation_episode
    measurements, sensor_data = client.read_data()
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 127, in read_data
    return pb_message, dict(x for x in self._read_sensor_data())
  File "C:\Users\lbai\Downloads\CARLA_0.8.2\PythonClient\carla\client.py", line 127, in <genexpr>
    return pb_message, dict(x for x in self._read_sensor_data())
RuntimeError: generator raised StopIteration

-----------------------------------------------------------------------
For compiled version 0.8.4, same error "ERROR: (localhost:2000) failed to connect: [WinError 10061] No connection could be made because the target machine actively refused it" appears again. This cannot be solved by using Example.CarlaSettings.ini.

closed 2019-04-17 by stale[bot]

freeclouds, 2020-11-06:
hi @linbaiwpi ,how to deal with this problem,i also meet this issue,thanks!

INFO:tensorflow:Restoring parameters from models/saveNet_1907.ckpt-158700
I1106 09:56:56.428069 140117888739136 saver.py:1284] Restoring parameters from models/saveNet_1907.ckpt-158700
Restoring from  models/
start to reset env
Traceback (most recent call last):
  File "/home/robot/work/code/rl/rl_CARLA/environment/carla/client.py", line 174, in _read_sensor_data
    raise StopIteration
StopIteration

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "test_ddpg.py", line 62, in <module>
    tf.app.run()
  File "/home/robot/.local/lib/python3.7/site-packages/tensorflow_core/python/platform/app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "/home/robot/.local/lib/python3.7/site-packages/absl/app.py", line 303, in run
    _run_main(main, args)
  File "/home/robot/.local/lib/python3.7/site-packages/absl/app.py", line 251, in _run_main
    sys.exit(main(argv))
  File "test_ddpg.py", line 47, in main
    state = env.reset()
  File "/home/robot/work/code/rl/rl_CARLA/environment/env.py", line 125, in reset
    observation, _, _, _ = self.step(action)
  File "/home/robot/work/code/rl/rl_CARLA/environment/env.py", line 67, in step
    measurements, sensor_data = self.client.read_data()
  File "/home/robot/work/code/rl/rl_CARLA/environment/carla/client.py", line 127, in read_data
    return pb_message, dict(x for x in self._read_sensor_data())
  File "/home/robot/work/code/rl/rl_CARLA/environment/carla/client.py", line 127, in <genexpr>
    return pb_message, dict(x for x in self._read_sensor_data())
RuntimeError: generator raised StopIteration

Heba-Syijari, 2023-10-28:
hi @linbaiwpi ,how to deal with this problem,i also meet this issue,thanks!