carla-simulator/carla #1231: Build fails with cmake

ykaganov, 2019-02-10:
Building fails with Clion based on the CMakeLists.txt file in the carla root directory. After running Setup.sh from /Utils/BuildToolsv to download all the dependencies. 
Running on Linux Ubuntu 16.04
CMAKE flags: -G Ninja -DCMAKE_BUILD_TYPE=Server

Part of the log is listed below: 

[ 95%] Building CXX object LibCarla/cmake/test/CMakeFiles/libcarla_test_server_release.dir/__/__/source/test/server/test_benchmark_streaming.cpp.o
[ 96%] Linking CXX executable libcarla_test_server_debug
CMakeFiles/libcarla_test_server_debug.dir/__/__/source/test/common/test_buffer.cpp.o: In function `testing::AssertionResult testing::internal::CmpHelperEQFailure<unsigned int, unsigned long>(char const*, char const*, unsigned int const&, unsigned long const&)':
/media/yan/Storage/GitRepos/carla/Build/googletest-1.8.0-ex-libcxx-install/include/gtest/gtest.h:1378: undefined reference to `testing::internal::EqFailure(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'

It seems I am getting undefined references all related to gtest. From what I understand, the header file and the compiled gtest object file do not seem to match. This is weird because using the makefile instead of cmake works just fine. I also ran Setup.sh before building which compiled gtest from scratch. I tried cleaning and rebuilding gtest but it does not help.

closed 2019-04-18 by stale[bot]

youngsend, 2020-08-24:
Hi, @ykaganov !
Have you succeeded in building CARLA with CLion?
If so, could you tell me how to do it?

dymons, 2021-03-07:
Hi, @youngsend and @ykaganov, I'm facing the same problem. The problem is due to the fact that no file was specified for the Toolchain.
```
-DCMAKE_TOOLCHAIN_FILE=${CARLA_ROOT}/Build/LibStdCppToolChain.cmake
```
In order to build a project, you need to call the following command in the command line (you can use the Makefile):
```
cd ${CARLA_ROOT}
make setup
mkdir build && cd build
cmake \
    -DCMAKE_GENERATOR="Unix Makefiles" \
    -DCMAKE_BUILD_TYPE=Client \
    -DLIBCARLA_BUILD_DEBUG=ON \
    -DLIBCARLA_BUILD_RELEASE=OFF \
    -DCMAKE_TOOLCHAIN_FILE=${CARLA_ROOT}/Build/LibStdCppToolChain.cmake \
    -DCMAKE_INSTALL_PREFIX=${CARLA_ROOT}/PythonAPI/carla/dependencies \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=1 \
    ..
make -j12
```
In order to build the project in the CLion, I did the following steps:
1. I modified the root one CMakeLists.txt as follows:
```
cmake_minimum_required(VERSION 3.5.1)
project(CARLA)

set(LIBCARLA_BUILD_DEBUG ON)
set(LIBCARLA_BUILD_RELEASE OFF)
set(CMAKE_BUILD_TYPE Client)
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)
set(CMAKE_INSTALL_PREFIX ${CMAKE_CURRENT_SOURCE_DIR}/PythonAPI/carla/dependencies)

include("Build/CMakeLists.txt.in")

add_subdirectory("LibCarla/cmake")
```

2. This is strange, but CMake ignores the CMAKE_TOOLCHAIN_FILE variable via initialization in CMakeLists.txt, as:
```
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_CURRENT_SOURCE_DIR}/Build/LibStdCppToolChain.cmake)
```
Therefore, we need to set this variable through the CLion settings: File - > Settings - > Build, Execution, Deployment - > CMake, and in the **CMake options** field, set the following for Debug and Release:
```
-DCMAKE_TOOLCHAIN_FILE=/path/to/carla/Build/LibStdCppToolChain.cmake
```
Unfortunately, CLion does not recognize the **${CARLA_ROOT}** environment variable, so you need to specify an absolute path.
After doing the above steps, the project in CLion should be built.