carla-simulator/carla #1234: Carla 0.9.2 : Vehicle control according to sensor data

krdindorkar19, 2019-02-11:


YashBansod, 2019-02-11:
@krdindorkar19 You can do this using the `self.world.on_tick(callback_function)` in the **World** Class.
Essentially, you just poll for the vehicle location, compute euclidean distance travelled since the previous location and maintain a sum of the eu_dist traversed.

krdindorkar19, 2019-02-12:
@YashBansod 
World class is in my 1st client (manual_control.py), I want to access sensor's(sensors which are spawned in client 1) data from client 2. So I need callback function to be called from client 2.
For distance I have used euclidean distance formula : 
>def get_distance_traveled(self):
>>	        '''
>>		this function returns distance vehicle traveled from starting point
>>		'''
>>		current_pos=self.ego_vehicle.get_location()
>>		distance = math.sqrt((current_pos.x-self.ego_vehicle_location.x)**2
>>						+(current_pos.y-self.ego_vehicle_location.y)**2
>>						+(current_pos.z-self.ego_vehicle_location.z)**2)
>>		return distance

closed 2019-02-18 by krdindorkar19