carla-simulator/carla #1240: [NoRenderingModeFixes] Needs more fixes

praveen-palanisamy, 2019-02-11:
[This issue is related to the the `NoRenderingModeFixes` branch]

I was trying to see if the `NoRenderingModeFixes` branch fixes issues with the no_rendering script in master. But, seems like it needs some more fixes.

https://github.com/carla-simulator/carla/blob/500904f1db90a81e6ed5f11585940f89b78139af/PythonAPI/no_rendering_mode.py#L471

Results in :
`TypeError: unsupported operand type(s) for *: 'float' and 'Vector3D'`

cc: @manishthani

nsubiron, 2019-02-11:
Hi @praveen-palanisamy,

That branch requires to recompile the client-side, we added a couple of bindings to Python API to simplify this sort of calculations.

You can download the egg file from the [latest build in Jenkins](http://158.109.9.218:8080/blue/organizations/jenkins/carla/detail/NoRenderingModeFixes/38/artifacts) and replace the one of 0.9.3 release for testing.

praveen-palanisamy, 2019-02-11:
Oops! I missed those changes. 
Thanks for the pointer.
I will close this issue.

closed 2019-02-11 by praveen-palanisamy