carla-simulator/carla #1251: [Fail] Running Carla in AWS

wendersonj, 2019-02-14:
Hello !
I'm trying to use the tutorial to carla_headless in the latest (0.9.3) in AWS machine, but everytime I get stucked in `sudo nohup Xorg :7 & ` it says that the command Xorg doens't exist ! Could anyone help me ?

PS.: Does anyone has a AMI ready for CARLA to use in AWS Machines ?

Last lines of log:
[image]

[Full Log](https://pastebin.com/7XYt7n2X)

_Originally posted by @wendersonj in https://github.com/carla-simulator/carla/issues/917#issuecomment-463131166_

wendersonj, 2019-02-21:
Yet I can connect to the EC2 with my machine, I was able to remote connect to a server with Carla.
My mistake was that i used port 2001 instead of **2000**.
Cheers !

wendersonj, 2019-02-22:
Just an update... I tried to connect to EC2 in a different network other than my University like my own... And I succeeded using port 2000. So, I'm closing this issue. 

Ps.: With manual_control.py, the graphics update from movements in Carla's client it's **very slow** ! 

Cheers !

closed 2019-02-22 by wendersonj

Kwanghoe, 2019-09-18:
Hello, @wendersonj .

I'm trying to following the[ "Running CARLA without Display and Selecting GPUs"](https://carla.readthedocs.io/en/latest/carla_headless/) (0.9.3)

Could you share your environment what you set up on you AWS EC2 instance?

Which version you used for this?
(instance type, graphic driver version)

#2102

harderthan, 2019-09-18:
Hello, @wendersonj .

I'm trying to following the[ "Running CARLA without Display and Selecting GPUs"](https://carla.readthedocs.io/en/latest/carla_headless/) (0.9.3)

Could you share your environment what you set up on you AWS EC2 instance?

Which version you used for this?
(instance type, graphic driver version)

#2102

wendersonj, 2019-09-18:
Hi @harderthan. Later today I will respond you properly.

harderthan, 2019-09-19:
@wendersonj, Sorry for hurry.
Is there any update?

wendersonj, 2019-09-19:
Hi! Sorry about the late.
Unfortunately I don't have the machine anymore, but the machine configuration was a machine with dedicated graphic card appropriately to run Carla. I had to install a lot of dependencies, but the one the I had most problems was some dependency related to graphic. I don't remeber properly, but I think it was Xorg, as I said in a previously post. Remember to verify your ports before connecting.

Sorry not being more helpful, but if you need anything else, please reach out.
Cheers.