carla-simulator/carla #1253: [ERROR] carla compiled version 0.9.3

Johnnyswong, 2019-02-15:
After I started the simulator, I got the following error:

LogShaders: Error: The global shader cache file data/carla_093-compiled/Engine/GlobalShaderCache-GLSL_150.bin is missing
xxxxxxx
Your application is built to load COOKED content No COOKED content was found his usually means you did not cook content for this build.
xxxxxxx
Exiting abnormally (error code: 1)

It seems that the content files is not correct, but after I put the content files in Content/carla, it's not work out yet.

nsubiron, 2019-02-15:
Following the installation instructions at [Running in a Docker](https://carla.readthedocs.io/en/latest/carla_docker/) and doing
```
docker pull carlasim/carla:0.9.3
docker run -p 2000-2001:2000-2001 --runtime=nvidia carlasim/carla:0.9.3
```
works for me

aayushARM, 2019-03-12:
@nsubiron I extracted the CARLA_0.9.4.tar.gz file and ran the sh file directly on an Ubuntu 18.04 machine(no Docker) with Tesla K80 and OpenGL 3, like this:

`./CarlaUE4.sh -opengl3`

But it gives below error:
[2019.03.12-02.37.21:730][  0]LogShaders: Error: The global shader cache file '/content/Engine/GlobalShaderCache-GLSL_150.bin' is missing.```

Your application is built to load COOKED content. No COOKED content was found; This usually means you did not cook content for this build.
It also may indicate missing cooked data for a shader platform(e.g., OpenGL under Windows): Make sure your platform's packaging settings include this Targeted RHI.

Alternatively build and run the UNCOOKED version instead.
```

Here's the full log: 
[log.txt](https://github.com/carla-simulator/carla/files/2955047/log.txt)

Any help is appreciated!

Johnnyswong, 2019-03-12:
I find that the cache file /Engine/GlobalShaderCache-GLSL_150.bin is missing, but there is a cache file of "/Engine/GlobalShaderCache-GLSL_430.bin". I wonder that is this bug related to the version of OpenGL?
@nsubiron

Johnnyswong, 2019-03-12:
I tried run the carla 0.9.4 compiled version on my local pc with OpenGL version 4.3, it works. 
So I guess this bug is because the 0.9.4 compiled version is compiled under OpenGL 4.3 ? 
Maybe you should try compiled the code under OpenGL 3.0?

aayushARM, 2019-03-12:
> I tried run the carla 0.9.4 compiled version on my local pc with OpenGL version 4.3, it works.
> So I guess this bug is because the 0.9.4 compiled version is compiled under OpenGL 4.3 ?
> Maybe you should try compiled the code under OpenGL 3.0?

That's exactly what I was wondering, but wanted some kind of confirmation. I think I'll have to compile for 3.0 and rerun. Will post any updates, thanks!

aayushARM, 2019-03-12:
> I tried run the carla 0.9.4 compiled version on my local pc with OpenGL version 4.3, it works.
> So I guess this bug is because the 0.9.4 compiled version is compiled under OpenGL 4.3 ?
> Maybe you should try compiled the code under OpenGL 3.0?

Hey I read the build instructions but don't know where to pass the -opengl3 flag, it is not mentioned in the instructions: [How to build on Linux](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/).
I don't want to launch the project in Unreal editor, just want the binaries generated by "make package" command but where do I say its for -opengl3?

MaxHBor, 2019-04-19:
May be it is not related to OpenGL version only.
I've build Carla from master-release on Windows 10. The error (with the cache file GlobalShaderCache-PCD3D_SM5.bin) appears with "Shipping" and "Development"  configurations.  Though it runs fine with "Development editor" or "Debug-editor configuration.

closed 2019-06-25 by stale[bot]

jediofgever, 2020-04-05:
run with opengl4

delenius, 2020-12-18:
I am also seeing this issue when running with the `-opengl3` flag, in a Linux VM/Mac OS host (I cannot run OpenGL 4 in this configuration), using the tar.gz bundle, release 0.9.9.