carla-simulator/carla #1254: Carla 0.9.2: How to get co-ordinates of lane using sensor.other.lane_detector

krdindorkar19, 2019-02-15:


aravindSwamy94, 2019-02-18:
Hi , I'm also looking for this issue, as I require lane coordinates either in world coordinate or local coordinate system. Just that the map is saved in opendrive format , but the lane coordinates are not provided. It would be better if we have lane coordinates for processing as well.

Unlingius, 2019-02-18:
As far as i know, at the moment it is not possible. But you can do it yourself using original opendrive File .xodr. There are list of road segments and list of lanes and markings for each segment. see https://github.com/carla-simulator/carla/issues/1082

YashBansod, 2019-02-20:
Instead of using `sensor.other.lane_detector`, I think you can easily implement this using the **Waypoint API** that works on `carla.Map`.

krdindorkar19, 2019-02-27:
@YashBansod can you explain how we will get lane's co-ordinated from waypoint api

YashBansod, 2019-02-27:
```python
# Get the carla.Map object from carla.World using
carla_map = world.get_map()
# Get the location of the ego vehicle using
ego_location = player.get_location()
# Get the waypoint on the lane closest to the ego vehicle using
waypoint = carla_map.get_waypoint(ego_location)
# Get the next waypoint(s) at particular distance using
waypoint.next(distance)
```

@krdindorkar19 you can always refer Carla's [Documentation](https://carla.readthedocs.io/en/latest/) and [Python API reference](https://carla.readthedocs.io/en/latest/python_api/) for full usage of their APIs.

aravindSwamy94, 2019-02-27:
@YashBansod: Few questions on this.

1. when there is an intersection ahead of you at a particular distance, waypoint.next(distance) gives you a list of points covering all the intersection lanes. in that case how will you choose a correct point, based on your route . Also my question is , is there any particular way , we can get the route of our ego vehicle in autopilot mode
2. Also this method gives us the lane center points, we have to calculate the lane boundary right and lane boundary left points for the particular lane. And while referring to #1082 where the lane for entire road is found and mapped with the segmented image. Is there a way we can do this with your approach by using carla.Waypoint Api. My understanding is that, to do this, we need to get a waypoint in each of the lane , in particular road and this has to be extended with waypoint.next(distance).

Please help on this regard.

YashBansod, 2019-02-27:
@aravindSwamy94 
1. Handling junction cases has been a problem for me too. I do not know of a way to know the route of the ego_vehicle in autopilot mode in advance. However, if you run the **automatic_control.py** example code provided in 0.9.3, they have implemented autopilot through the client side (Normally the autopilot is from the server side). In that implementation, they are plotting the route that the ego_vehicle is going to take (plotting is done using carla.DebugHelper, I think) and you will see the route plotted in advance at junctions too.   
  
If there was some way to get the topology of the route that the ego vehicle is following (or it is going to follow in the autopilot mode) , it would be great indeed. Unfortunately I do not know if this is already possible to do so with the current API. Maybe CARLA's members would know for sure.
  
2. To my understanding, the approach in #1082 is done by directly reading from the OpenDrive file of the map. @Unlingius was not using the Waypoint API for that. I think his explanation on his implementation makes it pretty clear on how he got those results. OpenDrive's representation of the lanes is a bit different and although I understand it, I have never tried to read from the OpenDrive files themselves. You can refer the OpenDrive's [Style Guide](http://www.opendrive.org/docs/OpenDRIVEStyleGuideRevC.pdf) for that.  
I would also request @Unlingius to share his code if possible.  
  
Regarding my suggestion on using the Waypoint API, the waypoints contain their transform (therefore their orientation is known) and the lane width is also provided. You can use this information to get the boundaries of the lane at that point. You can repeat this step for the next waypoint.

aravindSwamy94, 2019-02-27:
@YashBansod : 
  Thanks for your reply.

I would also like to have a look at @Unlingius code for this, if he is okay to share with us. 

Also, reading directly from openDrive file is one way to do this, which is more similar to a complete module like HDMap lane provider module. 
But can we get the lane information with carla API's 

I have few doubts regarding this, For example, consider my ego position is in road 71 and lane id as -2.
I can get the lane information for the same (road 71 and lane -2) by the method what u have suggested. But is there any way we can get the lane information for my neighbor lane(road 71, lane -1).  For this information, we need a waypoint from the same lane . Is there any possible way to get a waypoint of the subsequent lane, I came to know that this can be done by reading the openDrive file, but can we do this with any possible combinations of API ?

YashBansod, 2019-02-27:
@aravindSwamy94 I believe you can do that using the Carla's Waypoint API. I call tell you the pseudo code for the same:

```python
# 1. Get the carla.Map object from carla.World using
carla_map = world.get_map()  

# 2. Get all the waypoints in the map at say 1 meter distance
all_waypoints = carla_map.generate_waypoints(distance=1.0)  

# 3. Get the location of the ego_vehicle
location = player.get_location()  

# 4. Compute the the Waypoint closes to the ego_vehicle
nearest_waypoint = carla_map.get_waypoint(location, project_to_road=True)  

# 5. Find the closest waypoint to the `nearest_waypoint` in `all_waypoints`. Make sure the Road and Lane IDs are same.
waypoint_on_map = get_nearest_waypoint_same_lane(nearest_waypoint, all_waypoints)  

# 6. Find the closest waypoints to the adjacent lane of the `waypoint_on_map`. For this you narrow down your list to waypoints with same road id as `waypoint_on_map` but lane id equal to +1 or -1 of the lane id of `waypoint_on_map`.
adj_wayp_on_map_m1, adj_wayp_on_map_p1 = get_nearest_waypoint_adj_lanes(waypoint_on_map, all_waypoints)
```

I hope this helps. :penguin:

TDHTTTT, 2019-04-19:
> adj_wayp_on_map_m1, adj_wayp_on_map_p1 = get_nearest_waypoint_adj_lanes(waypoint_on_map, all_waypoints)

You can do:
```python
left_lane_waypoint = waypoint.get_left_lane()
right_lane_waypoint = waypoint.get_right_lane()
```

Note the it could return None if there is no adjacent lanes.

closed 2019-06-25 by stale[bot]

Pranav21091996, 2019-10-08:
@YashBansod @TDHTTTT  How do you calculate the deviation of the car from the centre of the road?

Ad4102, 2020-09-28:
> ```python
>
> # Get the carla.Map object from carla.World using
>
> carla_map = world.get_map()
>
> # Get the location of the ego vehicle using
>
> ego_location = player.get_location()
>
> # Get the waypoint on [...]

Hello @YashBansod . Is it possible to spawn these points back in the simulator? I have some problem while dealing with waypoint.next transform.