carla-simulator/carla #1258: Ensure tick order

nsubiron, 2019-02-15:
Ensure Carla actors tick in the right order

1. Read and send episode state (WorldObserver)
2. Update all the actors (UE4 Actor tick)
3. Parse client requests (RPC server tick)

In sync mode, loop on 3 until a certain message is received.

Right now this order is not guaranteed, this can add a latency of one tick on a lot of operations.

YashBansod, 2019-02-19:
@nsubiron I implemented an **IMU + CAN sensor** using the `world.on_tick(callback_function)` routine as you had mentioned in #1029 . This callback_function implements the following:
```python
id = timestamp.frame_count
timestamp = timestamp.elapsed_seconds
transform = self._player.get_transform()
accel = self._player.get_acceleration()
vel = self._player.get_velocity()
ctrl = self._player.get_control()
```
stores them in a dictionary and writes them to a *.json* file every cycle.
However I notice that this synchronization bug causes a lot of problems.  Some Cycles the data is correct. But, every few cycles I get same position data for 2-3 cylces and then suddenly the values jump. This causes various problems.  

Is there a way to resolve this issue?  

Please tell me even if there is a hack to make sure that `world.on_tick(callback_function)` is always executed after the state of ego vehicle is updated.

YashBansod, 2019-02-19:
Also, I wanted to know if this synchronisation issue happens in the `sensor.listen(callback_function)` too?

commit 169361cb (2019-02-17): Fix #1258, add FCarlaEngine to ensure tick order
commit 99dbd1c1 (2019-02-23): Fix #1258, add FCarlaEngine to ensure tick order
commit 43ed758f (2019-02-28): Fix #1258, add FCarlaEngine to ensure tick order
closed 2019-02-28 by nsubiron