carla-simulator/carla #1259: Carla 0.9.3 - Traffic light state is buggy and inconsistent for

mallela, 2019-02-15:
Hi, I'm using Carla 0.9.3 on a Ubuntu 16.04 machine.

I am trying to obtain the traffic light info in for each time stamp. I have noticed the following issues with it
1. get_traffic_light_state() is Green although there is no signal ahead
2. get_traffic_light_state() changes abruptly at the stopline if signal is Red. So does is_at_traffic_light().
3. get_traffic_light() does not always return the traffic light actor ID. I thought I recognized a pattern where it return the actor ID when  get_traffic_light_state() is Red, except that is not the case always.

Obtaining the traffic light state is such a useful feature! But there seems to be a bug here.

Thanks,
Praneeta

overlapjho, 2019-02-17:
I experience this every time without modification on original script 

In the image the vehicle detects the traffic light far ahead and respond into it. the `_proximity_threshold` is set to 10 which is the default
[image]

mallela, 2019-02-18:
Hi @overlapjho,
I'm sorry, I didn't understand completely. Is there a parameter `_proximity_threshold` that I can set somewhere? If yes, where can I find it?

overlapjho, 2019-02-20:
@mallela  ya, `_proximity_threshold` is found at `roaming_agent.py` at line 24

closed 2019-04-28 by stale[bot]

kargarisaac, 2019-11-14:
@mallela @jmarrr 
I have the same problem. When I pass an intersection, the "is_at_traffic_light()" remains active and traffic_light.get_state() remains red. Do you have any solution?