carla-simulator/carla #1260: Carla Lidar saving issue

moh3th1, 2019-02-18:
I am trying to save Lidar data using the method `save_to_disk` in the `automatic_control.py` but I think that the lidar is saved **upside down**. I am using MeshLab to visualize the lidar.

[image]

nsubiron, 2019-02-19:
I also noticed this, this has always been like this with our Lidar. I always wondered if this is a bug or a problem with the default axis visualization in meshlab.

tatttto, 2019-04-11:
Hi, @moh3th1 

Could you please show how to use `save_to_disk` function in the python script to save raw data for Lidar sensor? I tried this line:
`lidar.listen(lambda image: image.save_to_disk('_lidarout'))`

but it came out with this warning:

>  sensor object went out of the scope but the sensor is still alive in the simulation.

luigui2906, 2019-07-04:
To save lidar point clouds you have to use this:

`self.sensor.listen(lambda LidarMeasurement: LidarMeasurement.save_to_disk('_out/%06d.ply' % LidarMeasurement.frame_number))`

closed 2020-03-19 by germanros1987

visakhchikku, 2023-10-05:
Is it possible to save lidar data as .pcap file using carla?