carla-simulator/carla #1261: Adding new vehicle model - Auto

YDSh510, 2019-02-18:
Hi Carla Team.

I want to add my "Auto"- vehicle model in carla simulation environment. 

want to know how it is done. could not understand much from the documentation . Need more help.

It says "
If you want a simpler way you might copy our "General4wheeledSkeleton" from our project, either by exporting it and copying it into your model or by creating your skelleton using the same bone names and orientation. Bind it to your vehicle model and choose it when importing your vehicle into the editor. This way you won't need to configure the animation, you might just use "General4wheeledAnimation" (step 3) You also won't need to configure the bone names for your wheels (Step 7. Be carefull, you'll still need to asign the wheel blueprints)."

Where will I get existing General4wheeledSkeleton and how will I bind it to my vehicle model. Do we have any video for the same.

Thanks...

YDSh510, 2019-02-20:
Hi Team 

I have to add a new vehicle model in carla simulation . need your support as new to It.

I just have .fbx vehicle file how can I proceed further . Carla documentation is not much clear.

felipecode, 2019-02-20:
Sorry man, CARLA team is extremelly busy with the CARLA challenge.

YDSh510, 2019-02-21:
Sorry for trouble but just a little pointer  can help me making progress. Thanks in advance

I have a 3 wheeled vehicle , when I am trying to import its .fbx as Skeletal Mesh, selecting FordMustangGt1966_skeleton  (which is 4 wheeled) , it gives error - 

------------------------------------------

****### - FAILED TO MERGE BONES:

 This could happen if significant hierarchical changes have been made
e.g. inserting a bone between nodes.
Would you like to regenerate the Skeleton from this mesh?

***WARNING: THIS MAY INVALIDATE OR REQUIRE RECOMPRESSION OF ANIMATION DATA.*******
--------------------------------------------------

Do Carla have any skeleton for 3- wheeled vehicle or any other alternative way in which I can add my 3 wheeled vehicle model as npc.

YDSh510, 2019-03-04:
@ Carla Team- Hope Carla challenge is over .

It would be really great if you can guide me with above issue as I am really struck .

Thanks in advance.

XGodina, 2019-03-06:
Hi @YDSh510 ,

I apologize for answering so late that we've been very busy. Currently, we do not have a system to support 3-wheeled vehicles. I will talk to our technical artist because he has created a system for bikes and bicycles maybe that could help you.

gaurish123, 2019-03-07:
Thanks XGodina. Waiting for the update from Carla team soon on this.

TheNihilisticRobot, 2019-03-07:
@gaurish123 
We're back! 
Adding a 3 wheeled vehicle will be a bit more difficult than following the readme step by step, because the readme describes the use case for 4-wheeled vehicles and 2. I'll elaborate more in a moment.

TheNihilisticRobot, 2019-03-07:
So, If you have a 3wheeled vehicle  you won't be able to use the 4-wheeled skeleton, it just wont link. you'll have a few options you could choose and most probably, you'll have to create your skeleton and your animation. 

For that you just need to follow the full readme on how to add vehicles. Jumping over the green box that is just a shortcut for when your vehicle fits with our vehicle skeletons. this way you'll import your vehicle, choose to create a new skeleton when importing, and then creating and assigning a new animation for this vehicle. 

Other option that you have is adding an additional 4th wheel bone and importing it using our skeleton as a base. This bone will not have geometry and you'll need to take it out of the vehicle blueprint that will automatically assign it a Wheel.

Atlunch, 2019-03-07:
Hello everyone,

I'm also interested into importing my own car. I followed the description and it feels like i did it right. But where can i set now the car name like "vehicle.ford.mustang" and connect it to the BP that I can spawn the car from scripts?

YDSh510, 2019-03-08:
@TheNihilisticRobot 

I created 3-wheeled skeleton for my vehicle and imported it in unreal as skeleton mesh.
I tried following all the steps mentioned under section "Adding a vehicle" - skipping green box.

After that I added my vehicle entry in Vehicle_Spawner (in 4-wheeled array) and Vehicle_Factory blueprint (Vehicle array). But with this unreal is crashing.

Can you help me with where I am going wrong.

[image]

[Auto_sk.zip](https://github.com/carla-simulator/carla/files/2944658/Auto_sk.zip)

[image]

TheNihilisticRobot, 2019-03-08:
Ok @YDSh510 . For what I see, Unreal won't allow you to have a vehicle of less than 4 wheels. You'll have to add a fourth one no matter what.
What is your vehicle wheel configuration? Two forward wheels and one rear? Two Rear wheels and one forward? Is the vehicle symmetrical?

TheNihilisticRobot, 2019-03-08:
Ok @YDSh510 . for what I see, You'll have to add a fourth wheel no matter what...

What is your vehicle wheel configuration like? Two rear and one forward wheels? Two forward and one Rear? Is the vehicle symmetrical?

YDSh510, 2019-03-08:
o  o
 "Is the vehicle symmetrical"....  yes , It's two rear wheel and 1 front wheel in center. (like     V  shape)
                                                                                                                                                  l
Doubts -                                                                                                                                   o
1. "Other option that you have is adding an additional 4th wheel bone and importing it using our skeleton as a base. "                                                                                                  o  o            o o
                       Does it mean I need to add two front bones for single front wheel    V        or    V
                        or 2 front wheels?                                                                                ll                ll
                                                                                                                                     o               oo

2.This bone will not have geometry and you'll need to take it out of the vehicle blueprint that will automatically assign it a Wheel .... Not clear can you explain bit more.

Thanks ....

YDSh510, 2019-03-08:
[image]

YDSh510, 2019-03-11:
@TheNihilisticRobot - Any suggestion on above queries?

TheNihilisticRobot, 2019-03-11:
Hi again. 

Let me start stating that making vehicles with a number of wheels different than the usual 4 in Unreal is a bit harder than it looks. I'll start by giving you a overview of your options, once you decide witch one to follow let me know if you need more information about it.

You could add a third rear wheel bone between the two rear wheels (if you name it like one of the front wheels but block its steer or make it smaller than the rear ones you could then use our vehicle skeleton and settings and wont need to set up everything from scratch) , but I don't know how the vehicle blueprint will work with only one front wheel (We've made some tests in the past and the results were a bit unstable). 

You could do something similar to what our bikes do (they actually have 4 physical wheels and the actual model wheels only copy their rotation). but this setup will take a bit more work from your side since you'll have to create a 5 wheels skeleton (The two rear wheels, the centre frontal wheel that won't have collision at all and two invisible front wheels that will be parallel to the rear ones) and manually set up its animation and vehicle blueprint.

chandeep11, 2019-03-21:
HI @TheNihilisticRobot ,
We have created our own 4 wheeled vehicle skeletion (3 wheeled auto with front 2 wheels placed really close together which looks like a single wheel) with bones in blender. After importing it to unreal, in physical asset we can't able to select the individual wheels/body and unable to create bound shapes i.e.,sphere for the wheels and also for the Main car body. Any suggestion on this that what could be went wrong ?

TheNihilisticRobot, 2019-03-26:
Hi @chandeep11 I have no experience on blender so I don't know how much I'll be able to help you here. Please, me a screenshot of the physics asset you have in unreal. so I can check what is going wrong with your model.

chandeep11, 2019-03-26:
Hi @TheNihilisticRobot ,
Me and @YDSh510 are working in the same team. What we have done so far is tried creating auto with 
         o     o
            V
this      | |   bone structure(1 front wheel having 2 bones). But this one failed(crashed the UE4) when we      
            o                 
tried to spawn it. 

Then we created a model with 4 wheels by placing the the front 2 wheels nearly close. After importing  this in UE4, we cannot able to draw/create bound shapes in Physics asset. I've attached the model of our skeleton for reference. 

Please have a look at this and kindly suggest for any modification/improvements.
[auto_temp.fbx.tar.gz](https://github.com/carla-simulator/carla/files/3008242/auto_temp.fbx.tar.gz)

chandeep11, 2019-04-05:
HI @TheNihilisticRobot and @XGodina ,
We are now successfully able to create our custom vehicle(Auto) overcoming the above faced problems.

On following the Carla's - adding custom vehicle/asset procedure, we are unable to execute the last step ,
i.e.,
"Test it, go to CarlaGameMode blueprint and change "Default Pawn Class" to the newly created car blueprint."

When we are changing the default pawn class to our custom vehicle in CarlaGameMode and starting the simulation, CarlaUE4 is throwing this(attachment) error and it is crashing!, but the inbuilt carla vehicles are working/pawning without any issues. We have tested our model in UE4 separately and its working fine. 

Do we need to add the input events for our vehicle in the vehicle blueprint in Carla also, in-order to make the vehicle work?
[3wheel5.4.19.3.fbx.zip](https://github.com/carla-simulator/carla/files/3047671/3wheel5.4.19.3.fbx.zip)

I've also attached our model. kindly help us with that why the error is coming.!
[image]

chandeep11, 2019-04-10:
Hi @TheNihilisticRobot and @XGodina ,

any update on this will be very useful forour further progress.!

TheNihilisticRobot, 2019-04-15:
Hi again!

I'm sorry, we were out for the UnrealFest2019.

You've found part of our readme that is currently deprecated: _"Test it, go to CarlaGameMode blueprint and change "Default Pawn Class" to the newly created car blueprint_." The game mode here mentioned is obsolete. What you need to do, once your vehicle is finished, is add it to our vehicle factory:(Game\Carla\Blueprint\Vehicles\VehicleFactory) at the end of the "vehicles" array. Then you may run manualcontrol.py with your vehicle to test it (Type help on your manual control console to see how)

closed 2019-06-21 by stale[bot]

a-jahani, 2019-11-26:
> Hi again!
>
> I'm sorry, we were out for the UnrealFest2019.
>
> You've found part of our readme that is currently deprecated: _"Test it, go to CarlaGameMode blueprint and change "Default Pawn Class" to [...]

That's true! Thank you for your comment! documentation should be updated! I couldn't find anything related to the "vehicle factory" there. 

Finally, I was able to add my new vehicle:)

golnazraja, 2022-06-29:
Hello Carla team,

I read this issue, and I think the issue is not solved for general cases.

For example, is it possible to add vehicle with Unified Robotics Description Format (URDF) to CARLA? or heavy machines with different control model than a car?

hashim19, 2022-07-01:
Hi all,
A great thread for adding a non-standard vehicle to carla. @TheNihilisticRobot  @chandeep11 @YDSh510  can you guys guide me on how to add the wheel blue prints in case of 3 wheeled vehicles. I have two front wheels and one rear wheel. The 4th wheel is at the center and is raised above the ground and is smaller in size. Should I name it one of rear right or rear left wheels and change the shape accordingly? Need your help in this.