carla-simulator/carla #1262: "make launch" failed

vvrs, 2019-02-18:
I have been following the documentation to build carla on linux machine (Ubuntu 16.04). I could successfully build unreal engine. But when I executed the command "make launch", it gave the following error. 

[15/82] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/opendrive/parser/pugixml/pugixml.cpp.o
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:80: recipe for target 'LibCarla.server' failed
make: *** [LibCarla.server] Error 1

Any help resolving this is very much appreciated.

I want to use carla unreal editor to create custom maps. I would like to know if it is necessary to build the simulator or can I use pre-compiled version to carla unreal editor?

Thank you

closed 2019-02-25 by vvrs

ravishk1, 2019-03-14:
Hello @vvrs , I am getting this error. You closed this issue! Did you find any working solution for this ?
Thanks!!!

Stan-lee-1229, 2019-09-14:
@ravishk1 have you solved it？I also met this problem