carla-simulator/carla #1264: GeoReference in OpenDrive file

marcgpuig, 2019-02-18:
We are currently parsing the `geoReference` tag with the following content:
```xml
<geoReference><![CDATA[+lat_0=4.9000000000000000e+1 +lon_0=8.0000000000000000e+0]]></geoReference>
```

But RoadRunner export it like this:
```xml
<geoReference><![CDATA[+proj=tmerc +lat_0=49 +lon_0=8 +k=1 +x_0=0 +y_0=0 +datum=WGS84 +units=m +geoidgrids=egm96_15.gtx +vunits=m +no_defs ]]></geoReference>
```

Would be great if we could support the 2 options.

fpasch, 2019-02-18:
I would expect, that the current parser is able to work with the RoadRunner variant. Could you give it a try?

marcgpuig, 2019-02-22:
@fpasch I opened this issue because I had problems while opening a file generated with RoadRunner and I had a log message complaining about the geoReference.
I thought you could know why.

PR #1304 (2019-02-27): fix parsing of OpenDrive geoReference exported by RoadRunner
#### Description

OpenDrive definition files, exported by RoadRunner contain a more comprehensive geoReference definition. The parser can now cope with that.

Fixes #1264

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1304)
closed 2019-02-27 by nsubiron