carla-simulator/carla #1266: CARLA binary v0.8.2 crashed on ubuntu16.04 server

fangchuan, 2019-02-18:
Hi, i downloaded the binary package v0.8.2, and extract it successfully.
When i ran the `./CarlaUE4.sh` , a message bos told me "it supports at least OpenGL 4.3", but my openGL version is 3.0. then i ran the `./CarlaUE4.sh -opengl3`, but it still crashed, with the below error information:
`[2019.02.18-13.40.29:270][  0]LogRenderer: Reallocating scene render targets to support 1024x768 Format 10 NumSamples 1 (Frame:1).
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(1153306823) SRandInit(1153306823).
LogTaskGraph: Started task graph with 4 named threads and 9 total threads with 1 sets of task threads.
LogInit: Build: ++UE4+Release-4.18-CL-0
LogInit: Engine Version: 4.18.3-0+++UE4+Release-4.18
LogInit: Compatible Engine Version: 4.18.0-0+++UE4+Release-4.18
LogInit: Net CL: 0
LogInit: Compiled (64-bit): Apr  5 2018 13:20:57
LogInit: Compiled with Clang: 3.9.1 (tags/RELEASE_391/rc2)
LogInit: Build Configuration: Shipping
LogInit: Branch Name: ++UE4+Release-4.18
LogInit: Command line:  -Abslog=/home/fc/toolkits/CARLA/CarlaUE4/Saved/Logs/CarlaUE4-CRC.log /home/fc/toolkits/CARLA/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-32373-0E7F6608200215120028003444BDE695/
LogInit: Base directory: /home/fc/toolkits/CARLA/Engine/Binaries/Linux/
LogInit: Installed Engine Build: 1
LogInit: Presizing for max 100000 objects, including 0 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
LogInit: Object subsystem initialized
[2019.02.18-13.40.32:117][  0]LogInit: Linux hardware info:
[2019.02.18-13.40.32:117][  0]LogInit:  - we are the first instance of this executable
[2019.02.18-13.40.32:117][  0]LogInit:  - this process' id (pid) is 32508, parent process' id (ppid) is 32373
[2019.02.18-13.40.32:117][  0]LogInit:  - we are not running under debugger
[2019.02.18-13.40.32:117][  0]LogInit:  - machine network name is 'test-1'
[2019.02.18-13.40.32:117][  0]LogInit:  - user name is 'fc' (fc)
[2019.02.18-13.40.32:117][  0]LogInit:  - we're logged in locally
[2019.02.18-13.40.32:117][  0]LogInit:  - we're running with rendering
[2019.02.18-13.40.32:117][  0]LogInit:  - CPU: GenuineIntel 'Intel Core Processor (Skylake)' (signature: 0x506E3)
[2019.02.18-13.40.32:117][  0]LogInit:  - Number of physical cores available for the process: 6
[2019.02.18-13.40.32:117][  0]LogInit:  - Number of logical cores available for the process: 6
[2019.02.18-13.40.32:117][  0]LogInit:  - Cache line size: 64
[2019.02.18-13.40.32:117][  0]LogInit:  - Memory allocator used: binned2
[2019.02.18-13.40.32:117][  0]LogLinux: Skipped benchmarking clocks because the engine is running in a standalone program mode - CLOCK_REALTIME will be used.
[2019.02.18-13.40.32:118][  0]LogInit: Linux-specific commandline switches:
[2019.02.18-13.40.32:118][  0]LogInit:  -nodwarf (currently OFF): suppress parsing of DWARF debug info (callstacks will be generated faster, but won't have line numbers)
[2019.02.18-13.40.32:118][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2019.02.18-13.40.32:118][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2019.02.18-13.40.32:118][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2019.02.18-13.40.32:118][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2019.02.18-13.40.32:118][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2019.02.18-13.40.32:118][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2019.02.18-13.40.32:118][  0]LogInit:  - Physical RAM available (not considering process quota): 16 GB (16040 MB, 16425024 KB, 16819224576 bytes)`

Is there anyone can help me?
My server's configration:
1. virtualized by openstack
2. ubuntu16.04 server, kernel 4.15
3. OpenGL:
`glxinfo | grep OpenGL
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:
`
Whats more, i dont have a screen to display, i thought is it that causing the start processing of Unreal Engine?

fangchuan, 2019-02-20:
@nsubiron hi, could you help me ?

jkewang, 2019-02-25:
"OpenGL renderer string: llvmpipe (LLVM 6.0, 256 bits)" 
Your GPU is under level 5, which is not supported by UE4.

fangchuan, 2019-02-25:
@jkewang 
do you mean that i need to update my nvidia-GPU driver? My current driver version is 384.145

jkewang, 2019-02-25:
> @jkewang
> do you mean that i need to update my nvidia-GPU driver? My current driver version is 384.145

It seems like you are still running with intel graphic card. If you have a Nvidia-GPU , you need to check in ubuntu "About this computer" -> "graphic" . If it shows "llvmpipe (LLVM 6.0, 256 bits)", you need to reinstall GPU driver, until it shows "GeForce GTX 1070/PCIe/SSE2" (my GPU).

closed 2019-05-03 by stale[bot]