carla-simulator/carla #1268: Question: Collecting Data using multiple clients

Gaduu, 2019-02-18:
Hi all,

Quick few questions regarding collecting data using CARLA

I'm new to CARLA, been skimming through the documentation for a bit today to see what CARLA has to offer. The simulator looks really impressive and so far it looks like there are plenty of opportunities. 

My aim currently is to be able to simulate various scenarios, meanwhile collecting the necessary data that I would need to do some fancy neural net stuff. I'll do my best to briefly explain what I'm looking for.

I want to be able to:
- Connect to the server with multiple clients and be able to control them (e.g. 2 x manual_control.py from 2 different devices)
- Use another (3rd) script that would connect to the server and collect the data from both clients
- The data I want to collect is: Camera Imgs, Semantic Seg Imgs, Depth Imgs, some info about the vehicles such as: current speed, throttle, brake etc, as well as the distance from client1 to client2.

In that case I would have two different clients, both of them would have required sensors attached to them and I would control both of them with two different devices.

My questions are:
Can I use a separate (3rd) python script to connect to the server whilst both clients are connected and controlled and:
- Find the clients/vehicles by some sort of ID/name?
- Then use the above to collect the data from both clients/vehicles simultaneously? 

Any support will be greatly appreciated.
Cheers,

nsubiron, 2019-02-19:
Hi @Gaduu,

Any client that connects to the simulator has access to every actor. You can either search by actor id, or set a role name when you spawn the vehicle and then search for that role name. In the case of manual_control.py, it already sets the "hero" role name to the main vehicle. If you have two manual_control scripts connected then you should find two "heros" in the list of actors.

```py
heroes = [actor for actor in world.get_actors() if actor.attributes['role_name'] == 'hero']
```

The only limitation I see is that only one client can subscribe to a given sensor. Thus in the 3rd script you would need to create new cameras and attach them to the heroes. Adding cameras is quite costly, cameras slow down the simulation significantly. I would consider recording data also in the 2 manual_control.py just to add as few cameras as possible.

ADulian, 2019-02-19:
@nsubiron  thanks a lot, I'll give it a go and see what I'll be able to get out of it. Also, wondering what is the actual bottleneck here when you add number of cameras, is it the simulator or the hardware? 

Tbf, as long as I'll be able to keep up the simulation at 10-15fps I'll be more than happy. 

Cheers

nsubiron, 2019-02-19:
The kind of cameras we use have a big overhead in Unreal Engine renderer, but the faster is your GPU hardware the faster will go the simulator.

ADulian, 2019-02-19:
Great, thanks again.

closed 2019-02-19 by ADulian

AOOOOOA, 2021-12-28:
> Hi @ADulian,
> 
> Any client that connects to the simulator has access to every actor. You can either search by actor id, or set a role name when you spawn the vehicle and then search for that role name. In the case of manual_control.py, it already sets the "hero" role name to the main vehicle. If you have two manual_control scripts connected then you should find two "heros" in the list of actors.
> 
> ```python
> heroes = [actor for actor in world.get_actors() if actor.attributes['role_name'] == 'hero']
> ```
> 
> The only limitation I see is that only one client can subscribe to a given sensor. Thus in the 3rd script you would need to create new cameras and attach them to the heroes. Adding cameras is quite costly, cameras slow down the simulation significantly. I would consider recording data also in the 2 manual_control.py just to add as few cameras as possible.

@nsubiron 
Hi nsubiron,

I'm using carla.0.9.13 on ubuntu 18. I also tried to run two manual_control.py and use a third program to access the generated data from the manual-controlled vehicle. 

But I have met some problems:
1. I printed all the actors in the world, and when there is only one manual_control.py running, it shows there are 4 lidars. I don't know why.
2.  How to get the lidar data from the vehicle actor "hero"? I looked through the Actor part in Python API documentation but didn't find a suitable method.
3. I tried to use the _actor.listen()_ function for the lidar in the world, but the information retrieve program can't last long and I got nothing. Could you please tell me how to get the generated information from the vehicle?

Thanks!