carla-simulator/carla #1269: autopilot vehicle drives off road if not enabled fast enough after spawning

analog-cbarber, 2019-02-18:
This is observed using my Mac build of the master branch and the python3 API
(see https://github.com/analog-garage/carla/tree/mac-build-0.9) while running
the game in PIE mode.

If I spawn a vehicle at the 23rd or 24th spawn point and set autopilot it appears to always drive off the road at the next curve for every vehicle I have tried.  I haven't tried every single spawn point but running the spawn_npc script results in generally reasonable behavior.

~~~python
>>> sp = map.get_spawn_points()[23]
>>> print(sp)
Transform(Location(x=10.1901, y=207.506, z=1.8431), Rotation(pitch=0, yaw=-0.142975, roll=0))
>>> car = world.try_spawn_actor(random.choice(bpl.filter('vehicle.*')), sp)
>>> car.set_autopilot()
~~~

If I instead use a function to spawn the vehicle and immediately set autopilot the car does not drive off the road, so there appears to be some sort of timing issue with setting autopilot.

nsubiron, 2019-02-19:
This is happening because we're currently using a workaround to reuse our autopilot with the new road layouts. Previously, our autopilot was following a 2D road map that tell vehicles the direction of the road at any point except intersections, at intersections we use "route planners" that assign a random route (a list of waypoints) to the vehicle that enters their trigger box. This is still working in Town01 and Town02.

But with the more complex roads that we started adding, generating this 2D road map was becoming too difficult so we instead filled the town with route planners. We generate these route planners based on the OpenDrive info, basically one for each road segment. The problem with the route planners is that they only assign a route when a vehicle enters the trigger box, otherwise the vehicle will go straight. Thus if you switch autopilot on, the car will go straight unless you're lucky enough to find the trigger box of a route planner.

[image]

To avoid this happening on spawn, we give a list of recommended spawn points, right on top of trigger boxes. This way autopilot will always work if you spawn at one of the recommended spots.

@marcgpuig is working on improving our autopilot to make it work directly on OpenDrive info without the need of route planners, hopefully will have one that is able to switch to autopilot anywhere soon.

analog-cbarber, 2019-02-19:
Thanks for the reply.

In this case, I did create the vehicles in spawn points reported by the map. I don't think they start moving until I turn on the autopilot, so why does it matter if there is a delay between spawning the vehicle and turning on the autopilot? Given your explanation I would expect it to still work.

In practice, this does not matter that much to me since I can easily enough just do the spawn and enable autopilot in a function.

nsubiron, 2019-02-19:
The trigger box gets triggered when a vehicle enters the box, but not if it's already in it. It would also not work if the vehicle is spawned already inside the box, that's why the spawn points are a meter or two over the ground. It's ugly, we know, I hope we fix it soon :man_shrugging:

analog-cbarber, 2019-02-19:
So the vehicle triggers the box because it drops into it from above! I understand.

Wouldn't it work just as well to put the spawn point a little bit back from the trigger box in the orientation of the vehicle so that the vehicle will be sure to hit it if it moves forward?

nsubiron, 2019-02-19:
That would work too, but it would require checks to avoid intersections, curves, etc.

analog-cbarber, 2019-02-19:
Probably not worth changing if you think that general autopilot will work soon. In the meantime, you might consider adding a note to the documentation.

germanros1987, 2020-03-19:
This should be solved now with the new Autopilot (Traffic Manager)

closed 2020-03-19 by germanros1987