carla-simulator/carla #1270: Python API for Carla 0.9.3 requires libpng16  incompatible with ROS

anshulpaigwar, 2019-02-19:
Python API for Carla 0.9.3 requires libpng16 while ROS and Gazebo still work with libpng12. When I installed libpng16 using `apt-get` it auto removes some important Gazebo and ROS files. I think Carla should also stick to libpng12 for a while. Is there any way to install both libpng16 and libpng12 ??

nsubiron, 2019-02-19:
This is a known issue, unfortunately we don't know how to fix this except by removing PNG support #924. The problem is that we use Boost GIL for handling images, and as it turns out is not compatible with libpng12.

jkewang, 2019-02-25:
I think you can reinstall ros-kinetic-gazebo* after installing Carla. I did this way and they both work well in my machine.

anshulpaigwar, 2019-02-26:
Thank you @jkewang this worked for me too.

wattsavolt, 2019-03-21:
Did you ever try FreeImage? Link to python wrapper: http://freeimagepy.sourceforge.net/

rohanb2018, 2019-10-28:
I actually found another solution that doesn't require reinstalling ROS. If you install the `libpng16-16` package (which only has the libpng runtime library files) instead of `libpng16-dev`, none of the Gazebo/ROS files will be removed. And it seems that the CARLA Python API works fine with only the runtime library files: I tried the `tutorial.py` script and it saved the PNG files without a hitch.

Hopefully this works for others as well.

closed 2020-03-19 by germanros1987