carla-simulator/carla #1279: Is it possible to connect Carla to matlab simulink?

kimiya66, 2019-02-21:


kimiya66, 2019-02-21:
I want to use Carla to create my environment and vehicles also setting lidar sensor.  then matlab will read these sensors and does calculation on them through block diagram...finally see the results of all process (output of matlab blocks) in Carla environment. Is it possible to connect Matlab simulator to Carla? As I saw you said about python in documentation but is it possible to connect other API?
Thank you

Sakrust, 2019-02-21:
You can use a Python for Matlab interface (https://de.mathworks.com/help/matlab/getting-started-with-python.html), connect via third party tools or use the LibCarla C++ Interface to connect to Simulink.

PranjaLBiswas27, 2019-03-03:
Is it possible to get track bounds, lateral deviation from center of the road, coordinates in the ego frame and similar data from CARLA interface to MATLAB? Also, is it possible to create custom tracks easily within CARLA? Please let me know.

Kind Regards
Pranjal Biswas

Sakrust, 2019-03-04:
Sure, this should all be possible. You can use the waypoint API to get your lane information, the actor API for coordinates and then use one of the above mentioned ways to get your data from python to Matlab.

Creating custom tracks is possible via RoadRunner quite simple.

closed 2019-05-10 by stale[bot]