carla-simulator/carla #1281: How to spawn a imported own car in Unreal Engine&Carla?

JackRen88, 2019-02-21:
Hello,everyone.
According to doc(How to add assets),I imported own car  into the Unreal Engine,but when I went to CarlaGameMode blueprint and change "Default Pawn Class" to the newly created car blueprint(sfcar),as the picture shows：
[image],and entered the Play button,my own car was't spawned,while other car was spawned.so,how to spawn a imported own car in Unreal Engine&Carla? Could I have a packaged version of Carla with my own car?

Atlunch, 2019-03-07:
Hello everyone,

I'm also interested into importing my own car. I followed the description and it feels like i did it right. 
I can drive and control the car using the "CarlaGameMode". But how can I spawn now my new car as a npc using the python scripts. For example: To spawn the Mustang you have to use the name like "vehicle.ford.mustang"? Where can I definded this root? I'm thinking that i have also to link the BP to this kind of "vehicle....." name.

Best regards
Atlunch

TheNihilisticRobot, 2019-03-08:
Hi @Atlunch 

First of all, for the vehicle to be available to the PythonAPI you'll have to add it to the VehicleFactory Blueprint. Open it and look for the local variable "Vehicles" Add an item to it, assign your vehicle and fill all of the blanks. Compile, save and you should be able to use this vehicle from the PythonAPI.

Atlunch, 2019-03-08:
Awesome! Thank you  @TheNihilisticRobot ... works like a charme.

But my wheels are not going round ...hmmmm

Atlunch, 2019-03-08:
I see that all car physics have an additional bone with the Name "Ford Mustang1996_wheels".
Is this bone important? And if yes, how can i add this bone to my car?

Atlunch, 2019-03-08:
Selected the wrong AnimBP.... thank you for all.

closed 2019-05-14 by stale[bot]

deepakreddy63, 2019-06-21:
HI ,
I have reached till spawning the vehicle in carla through pythonAPI by adding to vehiclefactory blueprint class.

I could move the vehicle in CarlaGamemode. I could not able to move the vehicle in manual_control.py. 
Am I missing something?

krdindorkar, 2019-11-08:
@Atlunch @TheNihilisticRobot 
Could you please guide me in details that how I will add an item (as you mentioned below)  to local variable "Vehicles" >  "GenerateDefinitions" Function

> Hi @Atlunch
> 
> First of all, for the vehicle to be available to the PythonAPI you'll have to add it to the VehicleFactory Blueprint. Open it and look for the local variable "Vehicles" into the "GenerateDefinitions" Function. Add an item to it, assign your vehicle and fill all of the blanks. Compile, save and you should be able to use this vehicle from the PythonAPI.