carla-simulator/carla #1283: ImportErro when run python scenario_runner.py

dyli96, 2019-02-21:
Hi, I want to use scenario_runner on carla.
I first export PYTHONPATH=$PYTHONPATH:$~/carla/PythonAPI/dist/carla-0.9.3.egg
Then python scenario_runner.py --scenario FollowLeadingVehicle
I always encounter the error:
 import carla
ImportError: No module named carla

I would really appreciate if anyone can help me with it, thanks in advance!

dyli96, 2019-02-21:
It is fixed now. Then I encounter new errors.

del SCENARIORUNNER
NameError: name 'SCENARIORUNNER' is not defined

geekhut, 2019-02-22:
export PYTHONPATH=$PYTHONPATH:/home/ctg-sa/CARLA_0.9.3/PythonAPI/carla-0.9.3-py2.7-linux-x86_64.egg:/home/ctg-sa/CARLA_0.9.3/PythonAPI

if you are on carla 0.9.3 and using python2.