carla-simulator/carla #1286: Issues with obstacle detector

nsubiron, 2019-02-22:
I think the current implementation is not working properly.

I have created a client classs using this obstacle that I want to use for "line of sight":

```
class LineOfSightSensor(object):
    def __init__(self, parent_actor, hud):
        self.sensor = None
        self._history = []
        self._parent = parent_actor
        self._hud = hud
        self._event_count = 0
        self.sensor_transform = carla.Transform(carla.Location(x=1.6, z=1.7), carla.Rotation(yaw=0)) # Put this sensor on the windshield of the car.
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.obstacle')
        bp.set_attribute('distance', '200')
        bp.set_attribute('hit_radius', '12')
        bp.set_attribute('only_dynamics', 'true')
        #bp.set_attribute('debug_linetrace', 'true')
        bp.set_attribute('sensor_tick', '1')
        self.sensor = world.spawn_actor(bp, self.sensor_transform, attach_to=self._parent)
        weak_self = weakref.ref(self)
        self.sensor.listen(lambda event: LineOfSightSensor._on_LOS(weak_self, event))

    @staticmethod
    def _on_LOS(weak_self, event):
        self = weak_self()
        if not self:
            return
        print (str(event.other_actor))
        if event.other_actor.type_id.startswith('vehicle.'):
            print ("Event %s, in line of sight with %s at distance %u" % (self._event_count, event.other_actor.type_id, event.distance))
            self._event_count += 1
```
I am assuming that specifying the distance as 200 is the "length" of the raycast and hit_radius is the "thickness" of the line.

The results that I am getting are confusing:

Here some output logs:

```
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=55, type=vehicle.nissan.patrol)
Event 0, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 1, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 2, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 3, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 4, in line of sight with vehicle.nissan.patrol at distance 1200_
```

with the 'only_dynamics' flag aren't I suppose to see only vehicles? why Do I have static.vegetation?

If I leave the 'debug_linetrace' flag uncommented the rendering on the client breaks. this is the output on the carla server:
```
[2019.02.21-19.27.40:351][531]LogCarla: Spawning actor 'sensor.other.obstacle'
[2019.02.21-19.27.40:610][533]LogCarla: Spawning actor 'sensor.camera.rgb'
[2019.02.21-19.27.40:610][533]g.TimeoutForBlockOnRenderFence = "300000"
[2019.02.21-19.27.40:790][535]LogRenderer: Reallocating scene render targets to support 1280x720 Format 10 NumSamples 1 (Frame:38535).
[2019.02.21-19.28.12:798][540]LogTaskGraph: Wa
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.21ms for RT to recieve broadcast and do processing.
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.27ms to wait for named thread broadcast.
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.48ms to broadcast.
rning: Task graph took 277.05ms for RT to recieve broadcast.
```

I think there is a render method missing so that we can activate the visualization of the debug linetrace.

_Originally posted by @ignacioalvmar in https://github.com/carla-simulator/carla/issues/1127#issuecomment-466134515_

nsubiron, 2019-02-22:
Some more observations:
- Somehow the ray-cast is always done from (0, 0, 0) towards X.
- The render doesn't break if the size of the sphere is smaller.
- Vegetation (BP_RepSpline) is always detected even if we mark the spline as static.

dnovillo-CVC, 2019-03-26:
The main problem with the raytrace is that... it is Unreal's, so something too big in debug mode will break it anyway, it is heavy in performance.

PR #1441 (2019-03-26): Dnovillo/obstacle detection bugfix

#### Description

Fix for #1286

Fixes #1286  

#### Where has this been tested?

  * **Platform(s):** Linux
  * **Python version(s):** ...
  * **Unreal Engine version(s):** 4.21

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1441)
closed 2019-03-26 by nsubiron