carla-simulator/carla #1290: Run carla on Nvidia Tesla GPU

MahdiCat, 2019-02-23:
We want to install carla on a server with NVIDIA Tesla P100. Has anyone ever succeeded? Do we need to install Grid Driver or other driver? And what are the legal driver versions?
It would be great if there was a tutorial for installing carla on P100.
Thank you very much.

closed 2019-05-01 by stale[bot]

qhaas, 2019-07-02:
Our colleagues have had success running CARLA 0.9.5 on a Tesla V100, the successor to the Tesla P100 using the [CARLA Docker image](https://hub.docker.com/r/carlasim/carla) provided by the CARLA team running via [Singularity](https://sylabs.io/singularity).  AFAIK, no GRID driver is needed when not running inside a VM on Linux.

As a sanity check, this is how you would run it via Singularity, [nvidia-docker](https://github.com/NVIDIA/nvidia-docker) will likely work as well (especially since the images were written specifically for Docker)
```
# 'import' the docker image into singularity
singularity pull docker://carlasim/carla:0.9.5

# Create scratch directories on the host (Singularity will assign ownership of /home/carla to root)
install -d CarlaUE4/Saved Engine/Saved

# Launch the command from above with select folders mounted into the scratch directories
singularity exec --nv -B $(pwd)/CarlaUE4/Saved:/home/carla/CarlaUE4/Saved,$(pwd)/Engine/Saved:/home/carla/Engine/Saved carla_0.9.5.sif /bin/bash /home/carla/CarlaUE4.sh /Game/Carla/Maps/Town01 -benchmark -carla-server -fps=10 -carla-no-hud
```

Kailiangdong, 2019-07-24:
chmod: changing permissions of '/home/carla/CarlaUE4/Binaries/Linux/CarlaUE4': Read-only file system
Increasing per-process limit of core file size to infinity.
sh: 1: xdg-user-dir: not found
LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.
LogPakFile: Registered encryption key '00000000000000000000000000000000': 0 pak files mounted, 0 remain pending
LogPlatformFile: Using cached read wrapper
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.422070
LogICUInternationalization: ICU TimeZone Detection - Raw Offset: +0:00, Platform Override: ''
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin AISupport
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin CryptoKeys
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin MeshEditor
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin VariantManagerContent
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin AppleImageUtils
LogPluginManager: Mounting plugin AppleVision
LogPluginManager: Mounting plugin BackChannel
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin GeometryCache
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin LinearTimecode
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WebMMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin NetcodeUnitTest
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin PerformanceMonitor
LogPluginManager: Mounting plugin LauncherChunkInstaller
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin EditableMesh
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin WebMMoviePlayer
LogPluginManager: Mounting plugin Carla
LogStreaming: Display: Took  0.179s to configure plugins.
LogFileManager: Warning: ReadFile failed: Count=0 BufferCount=55 Error=errno=21 (Is a directory)
LogSsl: Warning: Unable to find a cert bundle in any of known locations. TLS may not work.
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = false  - Libcurl will NOT verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - MaxHostConnections = 16  - Libcurl will limit the number of connections to a host
LogInit:  - LocalHostAddr = Default
LogInit:  - BufferSize = 65536
[... 324 lines cut ...]
Malloc Size=48848 LargeMemoryPoolOffset=248112 
[2019.07.24-14.42.54:480][  0]LogCore: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000010

[2019.07.24-14.42.54:480][  0]LogCore: Fatal error!

0x0000000007ad6308 CarlaUE4!UnknownFunction(0x78d6308)
0x0000000007aaed7e CarlaUE4!UnknownFunction(0x78aed7d)
0x0000000005c3744c CarlaUE4!_Z33Linux_PlatformCreateDummyGLWindowP22FPlatformOpenGLContext(+0x2b)
0x0000000005c3970d CarlaUE4!_Z18PlatformInitOpenGLv(+0x93c)
0x0000000005c581ca CarlaUE4!_ZN17FOpenGLDynamicRHIC1Ev(+0x2c9)
0x0000000005c57eec CarlaUE4!_ZN23FOpenGLDynamicRHIModule9CreateRHIEN16ERHIFeatureLevel4TypeE(+0x1b)
0x0000000005cee529 CarlaUE4!_Z24PlatformCreateDynamicRHIv(+0x418)
0x0000000005ced8c6 CarlaUE4!_Z7RHIInitb(+0x75)
0x0000000003ebfdca CarlaUE4!_ZN11FEngineLoop7PreInitEPKDs(+0x2c89)
0x0000000003ecb5f8 CarlaUE4!_Z11GuardedMainPKDs(+0x47)
0x000000000789acdb CarlaUE4!_Z14CommonUnixMainiPPcPFiPKDsE(+0x95a)
0x00007f4eb1c71830 libc.so.6!__libc_start_main(+0xef)
0x0000000003ebd029 CarlaUE4!UnknownFunction(0x3cbd028)

[2019.07.24-14.42.54:490][  0]LogExit: Executing StaticShutdownAfterError
[2019.07.24-14.42.54:500][  0]LogCore: Warning: Unable to statfs('/home/carla/CarlaUE4/Saved/Crashes/crashinfo-CarlaUE4-pid-11159-000323C608D71045363BF7D4BC4D17B0/CarlaUE4.log'): errno=2 (No such file or directory)
Malloc Size=35397 LargeMemoryPoolOffset=283525 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
/home/carla/CarlaUE4.sh: line 5: 11159 Segmentation fault      (core dumped) "$UE4_PROJECT_ROOT/CarlaUE4/Binaries/Linux/CarlaUE4" CarlaUE4 $@

Kailiangdong, 2019-07-24:
[log.txt](https://github.com/carla-simulator/carla/files/3427365/log.txt)
@qhaas 
My gpu is p100.Thank you.I follow you method. But when I execute the last step, it crashs and not work.
Can you help me to check the log and find the fault.
Thank you

qhaas, 2019-07-24:
I was getting a similar error on one of our systems, not sure what fixed it... either a system restart or a CUDA process no longer sharing the same GPU as CARLA (UE4).  CARLA (0.9.5 at least) is hardwired, it seems, to use GPU 0, so verify that there is no other CUDA job (Type C) running on GPU 0 with `nvidia-smi -i 0`

Also, make sure Singularity is up-to-date, outdated versions seem to develop strange issues.

Kailiangdong, 2019-07-24:
@qhaas Thank you.
How to set gpu in this command ?
"singularity exec --nv -B $(pwd)/CarlaUE4/Saved:/home/carla/CarlaUE4/Saved,$(pwd)/Engine/Saved:/home/carla/Engine/Saved carla_0.9.5.sif /bin/bash /home/carla/CarlaUE4.sh /Game/Carla/Maps/Town01 -benchmark -carla-server -fps=10 -carla-no-hud"

Kailiangdong, 2019-07-24:
@qhaas 
I struggled in Carla for three days. May I ask why you use Docker to run Carla? Is it because Tesla(v100,p100)? 
I am a fresher and at first I didn't use docker and meet problem with opengl and nvidia driver.

qhaas, 2019-07-24:
We haven't figured out a way to select a specific GPU in Singularity or outside containers without root or SLURM... there is a sub-discussion about how to select a GPU other than GPU 0 in #225 

We use Singularity to run CARLA's Docker image because we don't have root access to the system with the V100s installed and to avoid dependency hell.

Kailiangdong, 2019-07-25:
@qhaas Thank you very much.
I don't use docker finally, but upgrade nvidia Gpu driver  to 410. Carla can run. But from xorg I can't see anything. 
Do you also follow this tutorial for remote display?
https://carla.readthedocs.io/en/latest/carla_headless/

Kailiangdong, 2019-07-30:
@qhaas Hi. Could you tell me your version of Opengl  which successfully run Carla? Thank you.

qhaas, 2019-09-21:
Tested with CARLA 0.9.6 (compiled version from Vendor), CentOS 7.7:
```
glxinfo | grep 'OpenGL version'
OpenGL version string: 3.0.0 NVIDIA 430.50
```