carla-simulator/carla #1291: RGB Camera in low quality mode (Carla 0.9.3)

nonsingularmatrix, 2019-02-23:
Hi, 
is there any way to use RGB-cameras when simulating in "low" quality mode?
Saved images of a camera only contain useful content when I'm using "epic" mode.

Thanks,
nonsingularmatrix

jagdishbhanushali, 2019-03-06:
I also saw this issue. When running simulator in low quality, camera saves garbage values whereas while running on epic mode, it save images perfectly fine with lots of lag.

germanros1987, 2019-03-06:
Hi, Could you be more specific? What do you mean by "useful" content?

nonsingularmatrix, 2019-03-06:
@germanros1987 
When I run the simulation in "low" quality mode, the saved image is not nearly similar to the one you receive when running in "epic" mode. As far as I remember, the image was just white or black.

germanros1987, 2019-03-06:
Please, report the code you are using for this experiment. Something that can reproduce the problem.

closed 2019-05-12 by stale[bot]