carla-simulator/carla #1292: No Shadow in Client (Carla 0.9.3)?

heshameraqi, 2019-02-24:
I know CARLA graphics are decent, I might be doing something wrong. I'm running CARLA 0.9.3 with "-quality-level=EPIC", and in the client window I can't see any vehicles shadows, and the graphics quality seems not good as in the image below.
For the client, I create a `pygame.display` and `blit(_surface)` to it, where `_surface` set via `pygame.surfarray.make_surface` with a `carla.ColorConverter.Raw` of `sensor.camera.rgb` sensor.

[image]

heshameraqi, 2019-03-03:
This does not happen anymore with CARLA recent 0.9.4 version. It should be closed.

closed 2019-03-03 by heshameraqi

soulslicer, 2019-08-20:
How did you fix this? I was messing around with some settings, and even with EPIC, it still lacks shadows. It was working before this

Jack-Pinnington, 2020-02-17:
Hello, I'm running on 0.9.6 and my Carla installation doesn't seem to have any shadows! What is the fix for this behaviour? [image]

Jack-Pinnington, 2020-02-17:
Hello,

I applied the fix here:
https://github.com/carla-simulator/carla/issues/2019

And my shadows have returned.

Thanks,
Jack