carla-simulator/carla #1293: carla.Actor.is_alive seems not dependable

praveen-palanisamy, 2019-02-24:
The following code snippet:
```python
for a in world.get_actors().filter("vehicle*"):
    if a.is_alive:
        a.destroy()
```
Results in:
`ERROR: unable to destroy actor: not found` 
 Indicating that the some of the `ActorView` is not valid.

If this won't work, what is the reliable way to destroy dynamic actors/objects?

nsubiron, 2019-02-25:
`is_alive` only checks if the actor instance has successfully called `destroy()`, it doesn't check if the actor is alive in the simulator, that would be costly and still not very useful because nothing guarantees someone else destroyed the actor in between `if a.is_alive` and `a.destroy()`.

In this case, AFAIK, for every actor returned by `world.get_actors()`, `is_alive` will always be true because the actor was alive in the simulator the last tick. I'm not getting the error you reported just by running that snippet, I think something else is also destroying the actors.

YashBansod, 2019-02-26:
@praveen-palanisamy the error occurred for me when I accidentally programmed multiple world.destroy() in a single episode.  
  
However, the error is just a log message in the terminal and doesn't effect he program flow. Therefore, as long as you have called agent.destroy(), you can be sure that it will be destroyed within some time.

DanielsJusts, 2019-04-10:
Hello!

I ran into the same issue described here that I relied on `actor.is_alive` to check whether the actor is still alive in the simulator.
So should I always use `world.get_actors()` and check if the actor still exists by its id or there is another more appropriate way to look at this issue?

Thanks!

nsubiron, 2019-04-15:
Hi @DanielsJusts,

Yes, I think the best way to check if an actor is still alive is `get_actors`, note that is more efficient if you provide the ids that you're interested in
```py
actors = world.get_actors(actor_ids=[...])
```

Destroying an actor that is already dead is actually safe, as a workaround to avoid getting the warning you could destroy the actors in bulk, this is executed asynchronously and doesn't care if an actor is not found
```py
client.apply_batch([carla.command.DestroyActor(x) for x in actor_ids])
```

DanielsJusts, 2019-04-16:
Thank you for your comprehensive and useful reply, @nsubiron! Keep up the good work! :)

closed 2019-06-22 by stale[bot]