carla-simulator/carla #1294: How to enable map mode or hero mode in 0.9.3?

s7ev3n, 2019-02-25:
Hi,
I saw there is a new feature "No rendering mode", and map mode or hero mode seems very interesting. However, I do not know or find any documentation on how to enable this mode.
Following the blog of [0.9.3](http://carla.org/2019/01/31/release-0.9.3/), I tried to set `DisableRendering=true` and start `manual_control.py`, however the pygame screen is black with some random colored lines in it.

Thx in advance.

silgon, 2019-02-25:
You'll find the file `carla/PythonAPI/no_rendering_mode.py`. I thought as first that it was really not rendering, but as far as I've seen, it's not rendering only for the client side. The python file I mentioned will give you the visualization as you saw it in the video.

s7ev3n, 2019-02-26:
@silgon Thanks so much!

s7ev3n, 2019-02-26:
Hi, @silgon 
I tried to start carla setting `DisableRendering=true`, and `no_rendering_mode.py`, it reported an error:
```
File "no_rendering_mode.py", line 469, in lateral_shift
return transform.location + shift * transform.get_forward_vector()
TypeError: unsupported operand type(s) for *: 'float' and 'Vector3D'

```
Have you met this issue? thx

silgon, 2019-02-26:
I haven't personally worked with property `DisableRendering=true` (I use the standard settings from the docker containers). Let's wait for a maintainer to respond. However if you only want not to display it (as it is my case) there is the  issue https://github.com/carla-simulator/carla/issues/225 which might help.

YashBansod, 2019-02-26:
@s7ev3n I was unable to replicate the error you are mentioning. Are you sure you have not accidentally modified the code in **no_rendering_mode.py"?

s7ev3n, 2019-02-27:
@YashBansod Thx for your attention.

I am sure that I do not modify any code in `no_rendering_mode.py`, and this time I made a fresh pull of carla repo to ensure that. However, it still reported the same error. I also tried #1217 and it has been solved as I pull code today.

Correct me if my steps are wrong: 1.I first start carla in no rendering mode which I set `DisableRendering=true` in `CarlaSetting.ini` and load this config, 2.I start `manual_control.py` in `carla/PythonAPI`, this gives a black screen with some colored lines, but I can see the HUD info, which I can control the actor, 3.I start `carla/PythonAPI/no_rendering_mode.py`, it crashes and give the error above.

The error reported in `transform.location + shift * transform.get_forward_vector()` is weird that I can add `Location`from `transform.location` and `Vector3D` from `transform.get_forward_vector()` in a script I wrote myself without any error.

YashBansod, 2019-02-27:
@s7ev3n, I am working with the code provided with the 0.9.3 release. I was unable to replicate the error you are mentioning in that version. But maybe things might have changed which is causing this instability.
Sorry I couldn't be of any more help.

closed 2019-03-04 by s7ev3n

YDSh510, 2019-05-22:
@YashBansod  - I am using Carla 9.4 and in then I am able to run no_rendering_mode.py properly.

Just start "Carla server" , once up , run manual_control.py (optional) and  no_rendering_mode.py. There is no need to set any flag in*.ini file.