carla-simulator/carla #1296: why there are no any sensors which are attached to the vehicle rendered out?

liqi198786, 2019-02-25:
hello,

i'm making a self-driving test platform with CARLA.
i'm now working with CARLA 0.9.2. i just use the manualcontrol.py included in the realse version. i made some changes in the script that i added a lidar and attached it to the vehicle. the lidar has been added successfully and i could get the pointcloud from it. but in the game scene, i could not see the lidar being rendered.
is there any method i could see the sensors added to the vehicle?

thanks

YashBansod, 2019-02-26:
@liqi198786 in the **manualcontrol.py** the following docstirng is provided:
```python
"""
Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    TAB          : change sensor position
    `            : next sensor
    [1-9]        : change to sensor [1-9]
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit
"""
```
Which basically tells that you can change the sensor visualizations on the hud using the keys [1-9]. However, this only creates a new sensor using the camera manager (it creates lidar too) and visualizes that. To visualize the sensor you have added yourself, you can modify the code in the HUD to plot the point cloud generated by your lidar. (Similar to how its already implemented for lidar created using camera manager).

liqi198786, 2019-02-26:
hi @YashBansod,really thank you for your reply. in fact,i didn't mean the visualization for pointcloud data from the sensor, i mean the sensor itsself like a lidar or a camera. i'd like to know where i have placed it on the vehicle.
[image]
just as what shown in the picture above.

YashBansod, 2019-02-26:
@liqi198786 I do not think that the sensors have any physical body in the simulator. I think this question would then be better handled by CARLA's members.

liqi198786, 2019-02-26:
@YashBansod,thanks, so i would not be caught up in the issue. just wait. maybe they will add it in future versions.

jagdishbhanushali, 2019-03-06:
Hi @liqi198786 , Sensors do not have physical body as you mentioned in your image. But you can see where sensor is placed in carla simulator. For that please follow below steps.

[image]

1. Run carla simulator from Unreal editor.
2. Run manual_control.py.
3. Click on game window in Unreal editor
4. Press F8.
5. Go to World Outlier on the left side of Unreal editor and filter by your vehicle model. (My vehicle model is AudiA2. I  executed my client as manual_control.py --filter vehicle.audi.a2).
6. Expand your vehicle model and you will see sensor attached to your vehicle.
7. To see where it is with respect to vehicle, click on sensor in World Outlier and hover your mouse on game window and press F.

liqi198786, 2019-03-07:
hi,@jagdishbhanushali, thanks for your reply. the situation for me now is that i once compiled the 0.9.0 version(so i could launch the UE4editor of v0.9.0), but now i'm working on v0.9.2 with the released binary, and also have made some changes in the mannualcontrol.py script of v0.9.2. 
i just tried to start the mannualcontrol.py of v0.9.2 to connect the carla server launched from v0.9.0 UE4editor. unfortunately, it failed with the message 'RuntimeError: rpc::rpc_error during call in function get_episode_info'.
i'd like to know if there is any way to edit the uncompiled(myself) released binary map level(like v0.9.2) in the once compiled UE4editor(v0.9.0 for my case) just through changing the folder content or something?

jagdishbhanushali, 2019-03-07:
You can't run manual_control.py of v0.9.2 with 0.9.0 server. However you can run carla v0.9.0 in Unreal editor, with manual_control.py of v0.9.0.  You should be able to do your task.

closed 2019-05-13 by stale[bot]

ShanshanYOU, 2019-05-16:
> Hi @liqi198786 , Sensors do not have physical body as you mentioned in your image. But you can see where sensor is placed in carla simulator. For that please follow below steps.
>
> [image]
>
> 1. Run [...]

Hi @jagdishbhanushali,
Thanks for sharing all those steps. I've tried to follow it but just for the second step it was stoped. Python scripts were failed to launch because fo an runtime error : "RuntimeError: rpc::rpc_error during call in function version ". Any further information will be really welcomed ! 

Cheers !

yuasabe, 2022-10-01:
Hi @jagdishbhanushali,

I thought it would be nice if the LiDAR actor also has a physical body in the simulator so that we can visually confirm where we have placed it in the scene. And if we can build this version of Carla as a package and not having to open up Unreal every time would help as well.
Can you provide me some guidance in editing the current RayCastLidar to include a physical body? I'm assuming the current implementation is just source code, and in order for it to have a physical mesh in the environment, it needs to be edited a bit in Unreal.

Thanks in advance.