carla-simulator/carla #1297: How to get all actors? world.get_actors()

silgon, 2019-02-25:
While using manual control (`carla/PythonAPI/manual_control.py`) in version `0.9.3` I can see that when colliding with an object I can know the kind of object I'm colliding with. For example if I collide with a wall, I will read **Collision with: Wall**, or when I collide with a tree I'll read **Collision with: Vegetation**.

I would like to get these actors from the world. When I use:

```python
[d.type_id for d in world.get_actors()]
```

I will not get the type Wall or Vegetation. The only static actors I see from the list are: `'sensor.camera.depth', 'sensor.camera.rgb', 'sensor.camera.semantic_segmentation','sensor.lidar.ray_cast', 'sensor.other.collision', 'sensor.other.gnss', 'spectator', 'traffic.speed_limit.30', 'traffic.speed_limit.60', 'traffic.speed_limit.90', 'traffic.traffic_light'`

Is there a way to detect the other actors?

nsubiron, 2019-02-25:
Hi @silgon,

The collision sensor creates "fake" actors when it collides with something that is not an actor, this is so we can retrieve the semantic tags of the object we hit. But all those objects are not registered as Carla actors, it would be very expensive tracking the state of every object in the scene.

silgon, 2019-02-26:
Thanks for the reply @nsubiron.
I see, I thought there were more objects that were just not displayed in the information given by the API.
Is there way to get the information about the buildings, vegetation and so from carla API?, or is it necessary to do some c++ to interface with unreal engine (never done that before).
I ask because I want to know the objects (e.g. buildings) that are near the car and not necessarily using the segmentation camera.

whbcumtb, 2019-03-05:
Hi @nsubiron
     I am reading your reply commented Feb 25, 2019, and I have a question that how to register an object as Carla actors?

whbcumtb, 2019-03-05:
Hi @nsubiron
    I am reading your reply commented Feb 25, 2019 and I have a question that how can I registor an object as a Carla actor?

whbcumtb, 2019-03-06:
Is there some method to get all the vertices of an actor/model and the 2D image coordinates?

nsubiron, 2019-03-06:
@whbcumtb To register another (type of) actor you need to modify the simulator's code, for instance, [here we register all the static mesh objects that are movable](https://github.com/carla-simulator/carla/blob/2e18abb576ff53947c9b7299c3e1c71b5fdbaac7/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.cpp#L189-L203).

And for getting the vertices of an actor, you can use their bounding boxes, but right now only vehicles and pedestrians have bounding boxes defined.

whbcumtb, 2019-03-07:
Thank you very much! @nsubiron . 
[image]
As shown in the above figure, we imported a turn_right model into world scene as an actor and converted it into a blueprint. According to your reply, we can get this model's actor by the function `get_actors()` in the way of scripting, but we can not get the corresponding blueprint by the function `get_blueprint_library()`. There are some further issues we want to resolve, and can you give us some advice in detail?
1) we added a variable that labels the model's shape when edited the blueprint, and want to access this variable value by scripting, how to achieve it?
2) how can we get the 2D image coordinates of the red corner points in this figure and the corresponding 3D vertices coordinates?

closed 2019-05-13 by stale[bot]

n1try, 2019-07-03:
> @whbcumtb To register another (type of) actor you need to modify the simulator's code, for instance, [here we register all the static mesh objects that are [...]

Is there any plans to include bounding boxes for actors other than vehicles and walkers any time soon?

ronyshaji, 2022-11-18:
@whbcumtb Able to solve the issue ?