carla-simulator/carla #1298: use of generate_waypoint() method

aravindSwamy94, 2019-02-25:
Hi,
            I just wanted to know what is the use of generate_waypoint(distance) method in carla API?

it says, it will generate waypoints all over the map at a particular distance?
For example if i give:
........
world = client.get_world()
map = world.get_map()
map_waypoints = map.generate_waypoints(0.1) -------------------------(1)
..........
waypoint = map.get_waypoint(ego.get_location()) ------------------------(2)
.......

1) Will this generate waypoints for every 0.1m in the entire map?
2) based on (1) , will it get a waypoint that is very close at 0.1m apart. 

For the above questions, MY understanding is that Map already has a set of waypoints at a predefined distance. They basically refer to center points of each Lane.

I just clearly want to understand the use of generate_waypoint(distance) method

nsubiron, 2019-02-25:
Hi @aravindSwamy94,

Waypoints are generated on demand, the Map does not have any waypoint stored.

- (1) Generates waypoints all over the map at approximately 0.1 meters distance between them.
- (2) Generates a waypoint at the center of a lane closest to the location provided.

There is also `get_topology()`, that generates pairs of waypoints defining the topology of the map, i.e. for each lane on each road segment generates a pair of waypoints, first for "this" lane, and second for the successor lane on the next road segment. It is useful for planning routes, the output can be fed to libraries like `networkx`.

YashBansod, 2019-02-26:
@aravindSwamy94 its always useful to visualize things. When I first used the `map.generate_waypoints()`, I used it as:
```python
waypoint_list = map.generate_waypoints(2.0)
# function to extract transform values from the waypoints
transform_list = extract_transforms(waypoint_list)
# function to plot the positions in the transforms.
plot_waypoints(transform_list)
```
The result was something like this. (its quite self explanatory)
[image]

aravindSwamy94, 2019-02-26:
@YashBansod and @nsubiron : Thanks for your input. Really helped me

closed 2019-02-26 by aravindSwamy94

anna-srl, 2020-04-28:
Hi @YashBansod , could you please specify where to use the code you mentioned?

marcgpuig, 2020-04-28:
@anna-srl you have a documented example [here](https://github.com/marcgpuig/carla_py_clients/blob/master/map_plot.py).

anna-srl, 2020-04-28:
@marcgpuig Many thanks!

sunqixing-creat, 2024-04-30:
> @aravindSwamy94 its always useful to visualize things. When I first used the `map.generate_waypoints()`, I used it as:
> 
> ```python
> waypoint_list = map.generate_waypoints(2.0)
> # function to extract transform values from the waypoints
> transform_list = extract_transforms(waypoint_list)
> # function to plot the positions in the transforms.
> plot_waypoints(transform_list)
> ```
> 
> The result was something like this. (its quite self explanatory) [image]
> 
> Also #992 might prove to be useful to you.
Can you share you code with me?

Vishnu213, 2024-11-20:
> Hi @aravindSwamy94,
>
> Waypoints are generated on demand, the Map does not have any waypoint stored.
>
> * (1) Generates waypoints all over the map at approximately 0.1 meters distance between them.
> * [...]

Thanks for this discussion, it is very helpful. I see that when we ask for get_waypoint(location), we get the wapoint at the center of the lane closest to the location provided, but what is the precision? I mean will I get next wapoint 0.1 m away exactly?