carla-simulator/carla #1299: [0.9.3] OpenDriverActor GetLanesOffset bug?

hdy117, 2019-02-25:
I am trying to load my own .xodr file, and I got some errors.
I do not know if it is the .xodr file's problem or something else. 
I fix the problem by doing the following modificaton. You may check if it is necessary.

In LibCarla/road/MapBuilder.cpp, in function MapBuilder::ComputeLaneCenterOffset()
change code from :
          lane_offset = (*general_info)->GetLanesOffset().at(0).second;
to:
          if((*general_info)->GetLanesOffset().size() > 0)
          {
            lane_offset = (*general_info)->GetLanesOffset().at(0).second;
          }

hdy117, 2019-02-25:
Also, I try to use the official tool from http://www.opendrive.org/ to test my .xodr file. it seems that nothing is wrong

hdy117, 2019-02-27:
I have checked the .xodr file. there is not "laneOffset" definition. 
In "LaneParser.cpp", your code does not deal with this situation. 
so the lane_offset vector is empty: 
struct Lanes {
    std::vector<LaneOffset> lane_offset;
    std::vector<LaneSection> lane_sections;
  };

marcgpuig, 2019-02-27:
Hi @hdy117, thanks for the report.

While expanding the waypoint API I encountered the same issue (and a couple more of the same style). 
I'm currently fixing several of them in my branch `marcgpuig/lane_change`.

hdy117, 2019-02-28:
> Hi @hdy117, thanks for the report.
> 
> While expanding the waypoint API I encountered the same issue (and a couple more of the same style).
> I'm currently fixing several of them in my branch `marcgpuig/lane_change`.

new problem:
[2019.02.27-14.00.16:589][336]LogOutputDevice: Warning: 

Script Stack (0 frames):

Ensure condition failed: NewTransform.IsValid() [File:/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp] [Line: 614] 

[2019.02.27-14.00.16:595][336]LogStats: FPlatformStackWalk::StackWalkAndDump -  0.006 s
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: === Handled ensure: ===
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: 
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: Ensure condition failed: NewTransform.IsValid() [File:/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp] [Line: 614]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: 
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: Stack: 
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcfb3c06 libUE4Editor-Engine.so!USceneComponent::InternalSetWorldLocationAndRotation(FVector, FQuat const&, bool, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp:2747]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcfb5886 libUE4Editor-Engine.so!USceneComponent::MoveComponentImpl(FVector const&, FQuat const&, bool, FHitResult*, EMoveComponentFlags, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp:2870]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcad26e8 libUE4Editor-Engine.so!AActor::SetActorLocation(FVector const&, bool, FHitResult*, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Actor.cpp:3362]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc046e0a974 libUE4Editor-Carla.so!AOpenDriveActor::BuildRoutes() [/home/hdy/HAD/Simulator/platform/carla-0.9.3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDriveActor.cpp:241]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc046e09a16 libUE4Editor-Carla.so!AOpenDriveActor::PostEditChangeProperty(FPropertyChangedEvent&) [/home/hdy/HAD/Simulator/platform/carla-0.9.3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDriveActor.cpp:98]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bfbf95d0 libUE4Editor-CoreUObject.so!UObject::PostEditChangeChainProperty(FPropertyChangedChainEvent&) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/CoreUObject/Private/UObject/Obj.cpp:445]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2ec57 libUE4Editor-PropertyEditor.so!FPropertyNode::NotifyPostChange(FPropertyChangedEvent&, FNotifyHook*) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyNode.cpp:2119]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2d2a7 libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(TArray<FObjectBaseAddress, FDefaultAllocator> const&, TArray<FString, FDefaultAllocator> const&, FPropertyNode*, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:543]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2b82b libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, FPropertyNode*, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:333]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2b29c libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:242]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf4bb9d libUE4Editor-PropertyEditor.so!FPropertyHandleBool::SetValue(bool const&, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:3432]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac052f0d libUE4Editor-PropertyEditor.so!SPropertyEditorBool::OnCheckStateChanged(ECheckBoxState) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/UserInterface/PropertyEditor/SPropertyEditorBool.cpp:79]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac0e6db2 libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, TTypeWrapper<void> (ECheckBoxState)>::Execute(ECheckBoxState) const [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:279]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac0e6e4c libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, void (ECheckBoxState)>::ExecuteIfSafe(ECheckBoxState) const [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:355]
[2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0b95b1241 libUE4Editor-Slate.so!SCheckBox::ToggleCheckedState() [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Slate/Private/Widgets/Input/SCheckBox.cpp:298]
[... cut ...]
[2019.02.27-14.00.34:043][749]LogPhysics: Warning: Initialising Body : Bad transform - Component: '/Game/Carla/Maps/GFTemple.GFTemple:PersistentLevel.RoutePlanner_144.TriggerVolume'  None
-nan,-nan,1393.368652|-0.232056,-101.229156,0.000000|1.000000,1.000000,1.000000
[2019.02.27-14.00.34:044][749]LogPhysics: Warning: Initialising Body : Bad transform - Component: '/Game/Carla/Maps/GFTemple.GFTemple:PersistentLevel.RoutePlanner_154.TriggerVolume'  None
-nan,-nan,1365.018799|0.202495,99.213562,0.000000|1.000000,1.000000,1.000000

hdy117, 2019-02-28:
> > Hi @hdy117, thanks for the report.
> > While expanding the waypoint API I encountered the same issue (and a couple more of the same style).
> > I'm currently fixing several of them in my branch `marcgpuig/lane_change`.
> 
> new problem:
> [2019.02.27-14.00.16:589][336]LogOutputDevice: Warning:
> 
> Script Stack (0 frames):
> 
> Ensure condition failed: NewTransform.IsValid() [File:/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp] [Line: 614]
> 
> [2019.02.27-14.00.16:595][336]LogStats: FPlatformStackWalk::StackWalkAndDump - 0.006 s
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: === Handled ensure: ===
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error:
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: Ensure condition failed: NewTransform.IsValid() [File:/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp] [Line: 614]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error:
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: Stack:
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcfb3c06 libUE4Editor-Engine.so!USceneComponent::InternalSetWorldLocationAndRotation(FVector, FQuat const&, bool, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp:2747]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcfb5886 libUE4Editor-Engine.so!USceneComponent::MoveComponentImpl(FVector const&, FQuat const&, bool, FHitResult*, EMoveComponentFlags, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Components/SceneComponent.cpp:2870]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bcad26e8 libUE4Editor-Engine.so!AActor::SetActorLocation(FVector const&, bool, FHitResult*, ETeleportType) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Engine/Private/Actor.cpp:3362]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc046e0a974 libUE4Editor-Carla.so!AOpenDriveActor::BuildRoutes() [/home/hdy/HAD/Simulator/platform/carla-0.9.3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDriveActor.cpp:241]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc046e09a16 libUE4Editor-Carla.so!AOpenDriveActor::PostEditChangeProperty(FPropertyChangedEvent&) [/home/hdy/HAD/Simulator/platform/carla-0.9.3/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/OpenDriveActor.cpp:98]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0bfbf95d0 libUE4Editor-CoreUObject.so!UObject::PostEditChangeChainProperty(FPropertyChangedChainEvent&) [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/CoreUObject/Private/UObject/Obj.cpp:445]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2ec57 libUE4Editor-PropertyEditor.so!FPropertyNode::NotifyPostChange(FPropertyChangedEvent&, FNotifyHook*) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyNode.cpp:2119]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2d2a7 libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(TArray<FObjectBaseAddress, FDefaultAllocator> const&, TArray<FString, FDefaultAllocator> const&, FPropertyNode*, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:543]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2b82b libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, FPropertyNode*, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:333]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf2b29c libUE4Editor-PropertyEditor.so!FPropertyValueImpl::ImportText(FString const&, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:242]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0abf4bb9d libUE4Editor-PropertyEditor.so!FPropertyHandleBool::SetValue(bool const&, unsigned int) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/PropertyHandleImpl.cpp:3432]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac052f0d libUE4Editor-PropertyEditor.so!SPropertyEditorBool::OnCheckStateChanged(ECheckBoxState) [/home/hdy/UnrealEngine_4.21/Engine/Source/Editor/PropertyEditor/Private/UserInterface/PropertyEditor/SPropertyEditorBool.cpp:79]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac0e6db2 libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, TTypeWrapper (ECheckBoxState)>::Execute(ECheckBoxState) const [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:279]
> [2019.02.27-14.00.16:595][336]LogOutputDevice: Error: [Callstack] 0x00007fc0ac0e6e4c libUE4Editor-PropertyEditor.so!TBaseSPMethodDelegateInstance<false, SPropertyEditorBool, (ESPMode)0, void (ECheckBoxState)>::ExecuteIfSafe(ECheckBoxState) const [/home/hdy/UnrealEngine_4.21/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:355]
[... cut ...]
> [2019.02.27-14.00.34:043][749]LogPhysics: Warning: Initialising Body : Bad transform - Component: '/Game/Carla/Maps/GFTemple.GFTemple:PersistentLevel.RoutePlanner_144.TriggerVolume' None
> -nan,-nan,1393.368652|-0.232056,-101.229156,0.000000|1.000000,1.000000,1.000000
> [2019.02.27-14.00.34:044][749]LogPhysics: Warning: Initialising Body : Bad transform - Component: '/Game/Carla/Maps/GFTemple.GFTemple:PersistentLevel.RoutePlanner_154.TriggerVolume' None
> -nan,-nan,1365.018799|0.202495,99.213562,0.000000|1.000000,1.000000,1.000000

hdy117, 2019-02-28:
> Hi @hdy117, thanks for the report.
> 
> While expanding the waypoint API I encountered the same issue (and a couple more of the same style).
> I'm currently fixing several of them in my branch `marcgpuig/lane_change`.

I export carla::opendrive::types::OpenDriveData into json, so I can check  OpenDriveData manually. 
Not everything in OpendriveData, just the junction information, road_link and lane_link information. If you need it, I can send it to you.

#include "nlohmann/json.hpp"
using json = nlohmann::json;

closed 2019-02-28 by hdy117