carla-simulator/carla #1303: simulator api version

HeYDwane3, 2019-02-25:
ubuntu 18.04
carla 0.9.3

when running the manual control mode, I got the following warning, and I am wondering how I can get the simulator API updated?

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.3 
WARNING: Simulator API version  = 0.9.2-131-g39f8e34b-dirty

YashBansod, 2019-02-26:
I think you should use the 0.9.3 carla simulator (pre-compiled package).   
From what I understand about the Warning Message, you are running the 0.9.2 simulator with 0.9.3 Client API.

slyer0009, 2019-02-27:
Have you called
`make PythonAPI`
in your Carla folder?

HeYDwane3, 2019-03-04:
> Have you called
> `make PythonAPI`
> in your Carla folder?

Hi, I have rebuild the 0.9.4, and also downloaded the pre-compiled version. This time, when I `make PythonAPI `, I got the err:
```
Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
Setup.sh: googletest-1.8.0-ex already installed.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client" configuration.
[1/4] Building CXX object LibCarla/cma...__/source/test/client/test_image.cpp.o
FAILED: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o 
/usr/bin/clang++-6.0  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_ENABLE_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_WITH_GTEST -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libstdcxx-install/include -I../googletest-1.8.0-ex-libstdcxx-install/include -I../../LibCarla/cmake/../source/test -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -std=c++14 -pthread -fPIC -Werror -Wall -Wextra   -O3 -DNDEBUG -MD -MT LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o -MF LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o.d -o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o -c ../../LibCarla/source/test/client/test_image.cpp
In file included from ../../LibCarla/source/test/client/test_image.cpp:10:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIO.h:9:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIOConfig.h:55:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png.hpp:11:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png/read.hpp:13:
../boost-1.69.0-install/include/boost/gil/extension/io/png/tags.hpp:25:14: fatal error: 'png.h' file not found
    #include <png.h>
             ^~~~~~~
1 error generated.
[2/4] Building CXX object LibCarla/cma...__/source/test/client/test_image.cpp.o
FAILED: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o 
/usr/bin/clang++-6.0  -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_ENABLE_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_WITH_GTEST -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libstdcxx-install/include -I../googletest-1.8.0-ex-libstdcxx-install/include -I../../LibCarla/cmake/../source/test -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -std=c++14 -pthread -fPIC -Werror -Wall -Wextra   -g -MD -MT LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o -MF LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o.d -o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o -c ../../LibCarla/source/test/client/test_image.cpp
In file included from ../../LibCarla/source/test/client/test_image.cpp:10:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIO.h:9:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIOConfig.h:55:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png.hpp:11:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png/read.hpp:13:
../boost-1.69.0-install/include/boost/gil/extension/io/png/tags.hpp:25:14: fatal error: 'png.h' file not found
    #include <png.h>
             ^~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:83: recipe for target 'LibCarla.client' failed
make: *** [LibCarla.client] Error 1
```
I have no idea what should I do there

lijecaru, 2019-03-06:
I got the same problem, I built Carla0.9.3 from source and I downloaded the precompiled Carla0.9.4. I can start Carla0.9.3 and I get the following error when typing `python manual_control.py`:

```
WARNING: Client API version     = 0.9.3 
WARNING: Simulator API version  = 0.9.3-51-ga0c2c21 

```
In Carla0.9.4 I get the following:
```
WARNING: Client API version     = 0.9.3 
WARNING: Simulator API version  = 0.9.4 

```

I have the impression it has to do with environmental variables, since I removed a version of anaconda I had and I removed it from my .bashrc file, and now I have this problem.

closed 2019-05-12 by stale[bot]

tn00364361, 2019-07-16:
I have a similar issue. I compiled the PythonAPI for Ubuntu 18.04 (python 3.6) using the following Dockerfile:

```
FROM ubuntu:18.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && \
    apt-get install -y wget software-properties-common git python3-distutils && \
    apt-get install -y build-essential clang-7 lld-7 g++-7 cmake ninja-build \
        python-dev python3-dev \
        libpng-dev libtiff5-dev libjpeg-dev tzdata curl unzip autoconf libtool rsync

RUN wget https://bootstrap.pypa.io/get-pip.py && \
    python2 get-pip.py --user --upgrade && \
    python3 get-pip.py --user --upgrade

RUN git clone https://github.com/carla-simulator/carla.git /carla
WORKDIR /carla
RUN git checkout 0.9.6

RUN update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-7/bin/clang 170 && \
    update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-7/bin/clang++ 170

RUN sed -i 's/bdist_egg/bdist_wheel/g' Util/BuildTools/BuildPythonAPI.sh && \
    make PythonAPI
```

Then I use `docker cp` to extract the wheel from a container and install the wheel using pip.

```
$ docker build -t carla_whl:0.9.6 PythonAPI
$ docker run -it --rm --name carla_whl carla_whl:0.9.6
$ docker cp carla_whl:/carla/PythonAPI/carla/dist/carla-0.9.6-cp36-cp36m-linux_x86_64.whl PythonAPI/carla/dist/
$ docker stop carla_whl

$ pip3 install --user PythonAPI/carla/dist/carla-0.9.6-cp36-cp36m-linux_x86_64.whl
Processing ./PythonAPI/carla/dist/carla-0.9.6-cp36-cp36m-linux_x86_64.whl
Installing collected packages: carla
Successfully installed carla-0.9.6
```

But I got the following warning when running `manual_control.py`.

```
$ ./PythonAPI/examples/manual_control.py
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down

    TAB          : change sensor position
    `            : next sensor
    [1-9]        : change to sensor [1-9]
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit

WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.6-dirty 
WARNING: Simulator API version  = 0.9.6
```

What is the difference between the "dirty" version and the regular one?

ronyshaji, 2022-09-09:
@tn00364361 Did you solved the error ?