carla-simulator/carla #1308: Is there a way to get the list of waypoints the vehicle will follow in the autopilot mode?

YashBansod, 2019-02-27:
We can get the Waypoint closest to the Ego_Vehicle and the Next Waypoint using the Waypoint API in Carla. However, it returns multiple Waypoints in case of intersections.  

Is there any way to know which Waypoint will be selected when the Ego_Vehicle is moving in Autopilot mode?

aroongta, 2019-02-27:
Route planners defined at the intersections give probabilities to each directions, so Waypoint of which direction will be selected is probabilistic and not deterministic. You can assign binary probabilities (0 or 1) to certain directions at intersections if you wanna be sure of the direction that the Ego vehicle needs to take.

aravindSwamy94, 2019-02-28:
Hi @aroongta : Thanks for your input, 
                           Where exactly is this calculation of probabilities , so that we can change this probability to 1 to make sure vehicle follows our defined direction. 
                              Because , I remember looking the code to a certain extent where they are using something like,
                          next_waypoint = random.choice(waypoint.next(distance))
which says that they are randomly choosing a waypoint, when they are approaching an intersection.

aroongta, 2019-02-28:
To give probabilities, u will have to change the values in the route planner assets in the UE editor.

aravindSwamy94, 2019-03-01:
@aroongta : Thaks for your reply,
               I m using the precompiled version of carla 0.9.3, and can you help me out, where exactly should i do this particular change.

aroongta, 2019-03-02:
Hi @aravindSwamy94 , I don't know if this change is possible in the pre-compiled version. This is because you need to changes the properties of the physical assets (route planner) through going in the Unreal Engine editor.

s7ev3n, 2019-03-28:
Hi, @aroongta
I am curious how `set_autopilot()` make a vehicle autonomous driving in carla. I did not find any code related to this. I can only find code in `./PythonAPI/agents/navigation` which implement global route planner and local planner. You mentioned a route planner assets in UE4 editor, it seems that the autopilot code is implemented there. Could you give me more insights about this ? Thanks very much.

-s7ev3n

s7ev3n, 2019-03-28:
@aroongta, I find the code, it is located in `Unreal/CarlaUE4/Plugins/Carla/Source/Carla`.
Thanks anyway :)

closed 2019-06-03 by stale[bot]