carla-simulator/carla #1311: multiple simulations crash (Carla 0.8.2)

Florence-C, 2019-02-27:
Hi , 

I'm trying to run multiple Carla clients with version 0.8.2. 
But everytime I run multiple server-clients (on different ports of course), I experience the same issue, all the simulators crashes with the same error : 

`ERROR: tcpserver 0 : connection failed: Operation canceled `
`[2019.02.27-09.22.44:736][ 96]LogCarlaServer: Warning: Failed to initialize, server needs restart` 
`ERROR: tcpserver 0 : error reading message: Bad file descriptor  `

I use the docker, and I tried to run all on one GPU, or on different ones, and I always have the same problem. 

Does anyone have the same issue ? 

Thanks !

zlw21gxy, 2019-02-28:
I face a similar issue #1307 ,althought I can running the simulator but it crash some time

nsubiron, 2019-03-01:
Couple of questions, just to be sure

- Are you trying to connect multiple clients to the same simulator instance? that's not available in version prior to 0.9.0.
- Are you taking into account that it uses 3 ports? 2000, 2001, 2002 by default, i.e. the port you provide and +1 and +2.

Florence-C, 2019-03-01:
No I'm only connecting one client to one simulator instance, and yes I took in account the different ports.

nsubiron, 2019-03-01:
Could it be that there is another process using the port?

Florence-C, 2019-03-06:
Sorry for the late reply.
No the ports used are only for Carla.

closed 2019-05-12 by stale[bot]