carla-simulator/carla #1316: Setting Throttle value

ADulian, 2019-02-28:
Hi,

I'm trying to to gather some data for training purposes. The problem I have is that having throttle set to 1 accelerates the vehicle too quickly, thus I was trying to change the initial throttle value that is sent to the vehicle via keyboard, however, changing throttle to anything below 1 breaks it to some extend, i.e. throttle = 0.5 will limit the car's speed to about 30km/h, 0.7 on the other hand will accelerate to about 60km/h but will somehow randomly decrease the speed every now and then.

Any clue how to deal with that matter?

Cheers

closed 2019-05-06 by stale[bot]

HaoZhouGT, 2021-02-16:
@ADulian Hi I found the exact same issue here, did you solve it later? Thanks

ADulian, 2021-02-17:
@HaoZhouGT No, sorry buddy.