carla-simulator/carla #1321: Using Routeplanner for an imported road map (fbx files)

dhruvsam, 2019-02-28:
I have imported a road model into CARLA. How do I use the routeplanner to make the cars drive on my road? How do I refine a spline for the routeplanners. I found the routeplanner object in CARLA but I do not know how to use it.

dhruvsam, 2019-03-08:
Hi Xavi,

Thanks for your reply. I am able to generate routes now! I am also trying out the roadrunner software and I am having some issues aligning the routes with the roads. After exporting the opendrive, fbx files and importing them into carla, I can see the routes generated but they are at a certain random offset from my map. Do you happen to have seen this issue before and what might be causing it?

XGodina, 2019-03-08:
Hi @dhruvsam 
Well, you have some ways to export maps from RoadRunner. One of them is using their Unreal pluguin. Another way is importing .Fbx and .oxdr and put in their correct folder. And the last way we have made a new system to export in the last release 0.9.4. Soon we'll update the Documentation. 

In your case, I thing you need rotate all meshes from RoadRunner 180º z channel. It is explained in our documentation. (https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/). But also, You can rotate in Unreal. Select all meshes from RoadRunner on Tab Details - Transform - Rotate 180º in z.

closed 2019-05-14 by stale[bot]

elnazbanani, 2023-04-03:
> Hi @dhruvsam Well, you have some ways to export maps from RoadRunner. One of them is using their Unreal pluguin. Another way is importing .Fbx and .oxdr and put in their correct folder. And the last [...]

Hi, I create a simple roade but It does not show any routeplanner. Therefore, I create in the unreal engine but it does not show the dot line between each boxes. what should I do?

elnazbanani, 2023-04-03:
> This issue has been automatically marked as stale because it has not had recent activity. It will be closed if no further activity occurs. Thank you for your contributions.

Hi, I create a simple roade but It does not show any routeplanner. Therefore, I create in the unreal engine but it does not show the dot line between each boxes. what should I do? could you help me?

elnazbanani, 2023-04-03:
> Hi, I create a simple roade but It does not show any routeplanner. Therefore, I create in the unreal engine but it does not show the dot line between each boxes. what should I do? could you help me?

Hi, I create a simple roade but It does not show any routeplanner. Therefore, I create in the unreal engine but it does not show the dot line between each boxes. what should I do? could you help me?

bg-sumu, 2024-02-23:
Hi @XGodina 
I have created a Route according to this tutorial, how to make sure my car follows that route ?

nikolatopic131, 2024-07-22:
> Hi @dhruvsam,
> 
> When you drop the route planner on the map you should assign a route. In Details Tab "Traffic Routes". You must assign a route by clicking on the plus. After that, deselecting the route planner and appear a spline. Now, you can move this spline and generate more points with Alt+LBM. Until you create the way you want.
> 
> [image]

Hello. Sorry for bringing back an old thread, but I'm having a problem with this step and I can't find a solution anywhere else. Basically I imported a map from Roadrunner into UE following CARLA's instructions (using make import) and even though the opendrive seemed fine in Roadrunner, routes weren't generated for my two on-ramps, so I want to add them manually. My problem is that doing what you described doesn't seem to work for me, I can't generate extra control points for my spline using alt+LMB and I checked the keyboard shortcuts in UE and can't find anything related to this. Moreover, for some reason there is a spline generated into the opposite direction (as shown in the picture), while I only want my route to go the way of the on-ramp. Also, even when I draw the spline as in the picture, once I de-select it, it disappears. Finally, the automatically generated routes in my map are red, while this manually made spline route looks white (not sure if this fact is of any significance, just curious what the difference is and why is it so). Do you perhaps know what the problem could be? I can't find the route planners mentioned anywhere else on the internet, and can't seem to fix this issue. Hence, I'd be very grateful if you could lend some insight.
[image]
[image]