carla-simulator/carla #1323: The destination of Basic Agent

hongyuli, 2019-02-28:
Hi,

I'm wondering what is the actual destination of a basic agent and how to set a destination.

In the automatic_control.py, the destination is set to Location(x=227.31, y=-5.08551, z=1.8431),  but the vehicle will always stop at Location(x=-77.8, y=24.2).

Is this a bug? Or is there any conversion between two locations?

(Using Carla 0.9.3, latest code from GitHub built from source, running on Ubuntu.)

slyer0009, 2019-03-04:
As far as i know when using the basic agent, you set the destination to any location. The agent will then find the nearest possible waypoint and set it as the true destination. Furthermore there is a factor of 10 between the worldcoordinates in the UE4Editor and in the PythonAPI locations (but that won't explain your coordinates). 
Maybe the vehicle will stop by the end of the RoutePlanner spline? And thats why the coordinates are so far apart.

hongyuli, 2019-03-13:
@slyer0009 Thanks for your response.

But, the destination in automatic_control.py is from the get_spawn_points(), which supposed to be a valid waypoint, right? So, I guess the vehicle should be navigated to that specific position.

Acan666, 2019-05-01:
> @slyer0009 Thanks for your response.
> 
> But, the destination in automatic_control.py is from the get_spawn_points(), which supposed to be a valid waypoint, right? So, I guess the vehicle should be navigated to that specific position.

Have you solved yet? I`ve found the same question. I`m thinking if there is a bug in the program.

hongyuli, 2019-05-01:
I think in their latest release 0.9.5, this problem is solved.

closed 2019-07-07 by stale[bot]