carla-simulator/carla #1325: BUG: Missing libcarla library

Tauvic, 2019-03-01:
CARLA 0.9.3 on Windows

In tried using the python client but when running spawn_npc.py I get a DLL load failure. I think a libcarla is required but not included in the Windows distribution.

https://drive.google.com/file/d/1YHTtYljaqv2kuvHYmqqtHSVMwdZaZAQe

nsubiron, 2019-03-01:
Yes, this is known issue, as a workaround you can compile (or download) a x64 `zlib.dll`, and paste it in the `PythonAPI` folder.

Tauvic, 2019-03-03:
I tried your advise, downloaded zlib.dll x64 and put it into the PythonAPI folder. But still no success.
I got this error:
--------------------------------------------------
Traceback (most recent call last):
  File "spawn_npc.py", line 24, in <module>
    import carla
  File "PythonAPI\carla-0.9.3-py3.6-win-amd64.egg\carla\__init__.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.6-win-amd64.egg\carla\libcarla.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.6-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
  File "C:\Users\tauvic.ritter\AppData\Local\Continuum\anaconda3\lib\imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: Kan opgegeven module niet vinden.
---------------------------------
Note that i renamed the egg because im using python 3.6.5 and not python 3.7
On line 7 in carla/PythonAPI/source/carla/__init__.py it says: **from .libcarla import ***

This makes me still wondering what is libcarla?

Tarek-Gh, 2019-03-04:
I am having the same issue
CARLA 0.9.4
Windows
Python 3.5.6

marcgpuig, 2019-03-07:
Try my branch `marcgpuig/zlib_static_link`

SAmmarAbbas, 2019-03-10:
Encountering the same error with python 3.7.2 using anaconda on Windows 10 with compiled version of CARLA 0.9.4

marcgpuig, 2019-03-12:
Great @SAmmarAbbas, thanks for sharing!

closed 2019-03-12 by marcgpuig

DanielDrde, 2019-03-28:
Hello,
i have the same Issue on Windows 10, Python 3.7.2 (64 Bit) and Carla 0.9.4.
Pasting the zlib.dll into the PythonAPI folder didn´t worked for me.
When there will be a compiled Version from Carla 0.9.4 for Windows which works?

SAmmarAbbas, 2019-03-28:
@DanielDrde it works for me on Windows 10 with Carla 0.9.4, so I guess it's not a fault on their part. Are you using anaconda? In any case, I would suggest creating a new virtual environment with Python 3.7, and testing it there.

DanielDrde, 2019-03-29:
I´ve tried it with Visual Studio Code and the Idle Editor with no success. But i haven´t tried to create a virtual environment. Do you use the virtualenv Package for creating this new environment? I´m not that much experienced with Python, so thanks for your Help!

SAmmarAbbas, 2019-03-30:
@DanielDrde I created a new environment with Anaconda for Python 3.7.2. I guess you could try with virtualenv as well.

DanielDrde, 2019-04-02:
I´ve tried it with a new virtual environment and a new installation of python 3.7.2, Carla 0.9.3 and 0.9.4 but unfortunately it didn´t worked. Same error Message:

Traceback (most recent call last):
  File "c:/Users/Server/Downloads/CARLA_0.9.3/dynamic_weather.py", line 28, in <module>
    import carla
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\__init__.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
  File "C:\Users\Server\AppData\Local\Programs\Python\Python37\lib\imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: Das angegebene Modul wurde nicht gefunden.

So i think i have to work with the stable Version of carla (0.8.4) and unfortunately without the API. 
Thanks for your help, Daniel.

fbuchinger, 2019-04-09:
Same problem for me using Carla 0.9.4, Python 3.7.3 under Windows 10.

DiegoOGV, 2019-04-29:
I have the same problem @fbuchinger but with windows 7. Did you fix it ?
Thanks

profmillerson, 2019-08-16:
I have the same problem

HanyangLiu, 2019-10-24:
same problem. But it only occurred to my own .py file. The same "No module named 'Carla'." didn't show when I run the built-in example .py files.

Xeth-Waxman, 2020-01-11:
Same problem here

hwacookie, 2020-06-18:
... and the same here with Windows and 0.9.9.

Tried with python 3.8 (which isn't supposed to work, as I found out later) and Python 2.7.18 (which should work, but doesn't)

Traceback (most recent call last):
  File "spawn_npc.py", line 24, in <module>
    import carla
ImportError: No module named carla

Hey, carla-team, we all know that linux is far more sexy and all, but couldn't you make this just a *little* bit easier for us poor Windows-victims? Pleaaaase?

Luca96, 2020-07-03:
My suggestion is to use `Python 3.7` along with a clean virtual environment. 

In particular, I downgraded Python from 3.8.0. to `3.7.8` and tested everything in a clean `virtualenv` environment managed by PyCharm.  That's worked for me on `CARLA 0.9.8` on Windows 10.

MustafaLotfi, 2021-02-20:
Hi, I exactly stuck on the same error. For me, it was because of the version of python (3.8) . So I installed python version 3.7 and problem fixed. (Carla 0.9.11)