carla-simulator/carla #1328: How to get the activation zone of a traffic light?

vvarano, 2019-03-01:
Hello,
I am new to Carla and I am trying to get the closest traffic light information using a vehicle location.
I can see in the UE4Editor how it is represented (shown in the image below).
Therefore, I would like to get the information of the traffic light activation area (the orange box zone in the image).

[image]

The information I would like to get are the position and the size of the activation area that is linked to the traffic light.
Is it possible to get these information from the PythonAPI somehow ?

germanros1987, 2019-03-03:
Hi,  we just added a simple way to do this un Python in our 0.9.4 release. Please, have a look at the code presented here: http://carla.org/2019/03/01/release-0.9.4/

Best regards.

closed 2019-05-09 by stale[bot]

NurbekHalikulov, 2020-12-01:
Hi @germanros1987,

is there a complete source code in github regarding the this problem? I did not understand what is going on in the code presented by you in the above.

Thanks

NurbekHalikulov, 2020-12-01:
Hi @germanros1987 , could you please tell what you mean with Util.length here in the given script:

s = Util.length(triggerBoundingBox.extent) + Util.length(vehicle.bounding_box.extent)

thanks a lot in advance