carla-simulator/carla #1329: How can I get the car model information ?

IreneLu12, 2019-03-01:
<Carla 0.9.2,  Built from source, Running on Ubuntu>

Hi,
I am trying to do some control research at Carla but I cannot find the car model information like vehicle mass, front wheel distance  and so on. And where is the definition of the  function 'ApplyControlToVehicle' ?  

Thanks  for your help.

IreneLu12, 2019-03-03:
Figured it out. Closed.

jkewang, 2019-03-15:
@IreneLu12 
I'm facing the same problem! how did you find the information of those vehicles? 
Thanks!

IreneLu12, 2019-03-19:
> @IreneLu12
> I'm facing the same problem! how did you find the information of those vehicles?
> Thanks!

Hi,
You can find the information in UE4 Editor. When you open the Carla Project in the editor, you can find all the vehicle physical model in Content/Carla/Blueprints/Vehicles.

jkewang, 2019-03-20:
Thanks, it helps me :)

closed 2019-05-26 by stale[bot]

cstamatiadis, 2019-06-26:
@IreneLu12  
Hi, 
I opened up the car asset within the editor, I found the tire specs. But I would also like to find the front wheel distance, but I don't seem to be able to find it anywhere. Could you help me with that?

cstamatiadis, 2019-06-26:
Never mind, after some additional digging I manged to find the info I needed.

friskerik, 2019-08-21:
I have problems finding this information. I can't seem to open up the project in UE4Editor (UE compiled fine but there seem to be some version issues with the Carla project version) so I tried to get it from the Carla API. 

I found the wheel base from the location of the wheels; looks OK. But how do I get the mapping from steer action ([-1, 1]) to angle at the wheels? There is a max_wheel_angle (about 70 deg on Audi TT which does not make sense to me).

So, my question is if someone can help me figure out how do I find the function 

delta = f(steer)

where delta is the angle at the wheels?

sinamoradi1989, 2022-04-26:
Hi,
Can someone provide me with the vehicle specifications of Tesla model3? I am doing my thesis on MPC control strategy in roundabouts for safe autonomous driving; however, I don't have access to UE4 (only binary but not GUI). I want to build the plant in Matlab and in the end, demonstrate my work in CARLA.
Idk if the specifications for the vehicles are listed anywhere publicly. I just didn't find it, and this page was the most useful so far.
Appreciate your help fellow researchers :)

[image]

fatemehjdi, 2022-12-01:
Hi,
Can someone provide me with the vehicle specifications of Tesla model3?

sinamoradi1989, 2022-12-08:
Hi there,

Your question is too general to answer. You can find some vehicle
specifications via the python API of Carla. A few others you may find on
the internet (like the frontal area, to calculate the drag force). But a
lot of others are hard to find such as the cornering stiffness of the front
tires.

Good luck,
Sina

fatemehjdi, 2022-12-08:
> Hi, Can someone provide me with the vehicle specifications of Tesla model3? I am doing my thesis on MPC control strategy in roundabouts for safe autonomous driving; however, I don't have access to [...]

Hey!
Did you accomplish doing your research with Tesla? I am curious to read your paper if you published any.

sinamoradi1989, 2022-12-09:
Hi,

The Paper is publicly available with the title "Comfort-based AI-driven
Roundabout Control for Automated Vehicles" by "S. Moradi, A. Nadar, J.
Härri" and after June when the MT-ITS 2023 conference is held.  You can
find a simulation in the link below. Yet, what you are looking for is only
a small section of that paper.
https://youtu.be/yiMYDcs57aU
Also, If you refer to the parameters in the table that I earlier posted,
they are just examples and do not help you to completely define the
dynamics. For instance, knowing the velocity and camera distance doesn't
make sense. However, as mentioned before, you can find the default values
of Carla for some of the Tesla Model 3 parameters using the Python API.
Search for the terms carla.VehiclePhysicsControl
<https://carla.readthedocs.io/en/latest/python_api/#carla.VehiclePhysicsControl>
 and carla.WheelPhysicsControl
<https://carla.readthedocs.io/en/latest/python_api/#carla.WheelPhysicsControl>
and
read the documentation. Then, you can for example print the mass, which is
m=1847 kg. It is worth mentioning that since we didn't come up with all of
the required values, we relied on system identification techniques to
estimate the vehicle dynamics. You may also do the same.
I hope this helps, and good luck with your research.

Best Regards,
Sina

yihsinma, 2024-05-09:
> > 大家好，這篇論文的標題是“基於舒適的人工智慧驅動的自動車輛環島控制”，作者是“S. Moradi、A. Nadar、J. Härri”，並於6 月MT-ITS 2023 會議召開後公開發布。您可以在下面的連結中找到模擬。然而，您要查找的只是論文的一小部分。https://youtu.be/yiMYDcs57aU另外，如果您參考我之前發布的表格中的參數，它們只是示例，並不能幫助您完全定義動態。例如，知道速度和攝影機距離是沒有意義的。但是，如前所述，您可以使用 Python API 找到某些 Tesla Model 3 參數的 Carla 預設值。搜尋術語 carla.VehiclePhysicsControl https://carla.readthedocs.io/en/latest/python_api/#carla.VehiclePhysicsControl和carla.WheelPhysicsControl [https://carla.readthedocs.io/en/latest/python_api/#carlala。](https://carla.readthedocs.io/en/latest/python_api/#carla.WheelPhysicsControl)閱讀文檔。然後，您可以列印質量，即 m=1847 kg。值得一提的是，由於我們沒有提出所有所需的值，因此我們依靠系統識別技術來估計車輛動力學。您也可以這樣做。我希望這對您有所幫助，祝您研究順利。最好的祝福，新浪
> > […](#) 
> 
> 你好，新浪， 我正在做一個類似的項目，它實現了 MPC，我需要計算出車輪的側偏剛度的值。正如你所提到的，我使用 Carla 的 Python API 來確定一些值，例如質量和轉動慣量等。一些幫助將不勝感激:) 另外，我找不到你提到的論文。可以留個連結嗎？非常感謝！

hello!
我要怎麼知道vehicle.tesla.model3的轉動慣量和側向剛性
我只知道轉動慣量是1kg*m2
有實際的值嗎

aalaeDev, 2025-01-21:
Hi, this reply is too late :) but I have just stumbled upon this topic while looking for the same thing. I added this repo that contains vehicles model information (Vehicle Physics Control) for all available Carla vehicles.