carla-simulator/carla #1331: Addition of RADAR to simulator.

pparmesh, 2019-03-03:
I am working on a project that requires fusion of RADAR and camera data. Since there is no RADAR in Carla I am currently thinking of extracting ground truth of all NPC vehicles within a FOV and adding some sort of noise to the data to mimic RADAR output.
Is it likely that RADAR will be added to the simulation environment in the near future ?

Sakrust, 2019-03-04:
Roadmap: https://github.com/carla-simulator/carla/blob/master/Docs/img/carla_timeline.png

closed 2019-05-10 by stale[bot]

ivannson, 2019-07-04:
@pparmesh Were you able to find a way to get RADAR output from Uneal?

pparmesh, 2019-07-04:
I tried to model one by getting ground truth data of vehicles within a certain field of view and then added noise on top of it.

dnychennnn, 2020-07-16:
@pparmesh Hey, would you like to share some thoughts on the noise modeling?