carla-simulator/carla #1333: stub files for python api

analog-cbarber, 2019-03-03:
It would be nice if the new python API included stub files so that type checking and code completion will work better in IDE's that support it (e.g. PyCharm).

PyCharm appears to be reading the doc strings, which currently contain C++ signatures, and gets confused about what the expected arguments are for methods in the Python API.

See https://www.python.org/dev/peps/pep-0484/#stub-files

Observed using the 0.9.4 API.

closed 2020-03-19 by germanros1987

HarryWindsor, 2020-04-16:
How can the PythonAPI documentation from #1715 and #1814 be used in PyCharm?

analog-cbarber, 2020-04-17:
I think the doc strings need to be pushed to stub files that are shipped with the package before PyCharm will be able to find them.

mathiaswold, 2022-09-14:
Hi, I created a repository containing stubs for the Python API: https://github.com/mathiaswold/carla-python-stubs