carla-simulator/carla #1334: Make PythonAPI issue

HeYDwane3, 2019-03-04:
Ubuntu 18.04
carla 0.9.4

After updating the carla to 0.9.4, and then `make PythonAPI`

I got the err:

```
Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
Setup.sh: googletest-1.8.0-ex already installed.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client" configuration.
[1/4] Building CXX object LibCarla/cma...__/source/test/client/test_image.cpp.o
FAILED: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o 
/usr/bin/clang++-6.0  -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_ENABLE_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_WITH_GTEST -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libstdcxx-install/include -I../googletest-1.8.0-ex-libstdcxx-install/include -I../../LibCarla/cmake/../source/test -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -std=c++14 -pthread -fPIC -Werror -Wall -Wextra   -O3 -DNDEBUG -MD -MT LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o -MF LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o.d -o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_release.dir/__/__/source/test/client/test_image.cpp.o -c ../../LibCarla/source/test/client/test_image.cpp
In file included from ../../LibCarla/source/test/client/test_image.cpp:10:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIO.h:9:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIOConfig.h:55:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png.hpp:11:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png/read.hpp:13:
../boost-1.69.0-install/include/boost/gil/extension/io/png/tags.hpp:25:14: fatal error: 'png.h' file not found
    #include <png.h>
             ^~~~~~~
1 error generated.
[2/4] Building CXX object LibCarla/cma...__/source/test/client/test_image.cpp.o
FAILED: LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o 
/usr/bin/clang++-6.0  -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_ENABLE_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_WITH_GTEST -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libstdcxx-install/include -I../googletest-1.8.0-ex-libstdcxx-install/include -I../../LibCarla/cmake/../source/test -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -std=c++14 -pthread -fPIC -Werror -Wall -Wextra   -g -MD -MT LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o -MF LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o.d -o LibCarla/cmake/test/CMakeFiles/libcarla_test_client_debug.dir/__/__/source/test/client/test_image.cpp.o -c ../../LibCarla/source/test/client/test_image.cpp
In file included from ../../LibCarla/source/test/client/test_image.cpp:10:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIO.h:9:
In file included from ../../LibCarla/cmake/../source/carla/image/ImageIOConfig.h:55:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png.hpp:11:
In file included from ../boost-1.69.0-install/include/boost/gil/extension/io/png/read.hpp:13:
../boost-1.69.0-install/include/boost/gil/extension/io/png/tags.hpp:25:14: fatal error: 'png.h' file not found
    #include <png.h>
             ^~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:83: recipe for target 'LibCarla.client' failed
make: *** [LibCarla.client] Error 1
```

Anybody know what should I do? It looks like something wrong with the LibCarla

MaskedName, 2019-03-07:
sudo apt install -y libpng-dev

should resolve the issue.

closed 2019-05-13 by stale[bot]

Logan-Purcell, 2019-08-16:
Im having the same issue

Faabdoly, 2019-09-30:
I have same issue on windows 10