carla-simulator/carla #1338: Carla Simulation from Point A to Point B

imran514, 2019-03-05:
Using Carla is it possible to simulate autonomous driving from point A to Point B on the map/town ??

Sakrust, 2019-03-05:
By using the Agent-Classes provided in the PythonAPI folder you should be able to navigate your vehicle from A to B.

lijecaru, 2019-03-05:
> By using the Agent-Classes provided in the PythonAPI folder you should be able to navigate your vehicle from A to B.

Is there an example on how to use them? I read the code but still failed to fully understand how to use them to navigate. Do I need to create a new python file or can I modify one of those to navigate around?

imran514, 2019-03-05:
Thank you @Sakrust 
@lijecaru in PythonAPI folder there is file called automatic_control.py, you can refer that file on how to use agent classes.

imran514, 2019-03-06:
@Sakrust 
when i  try running automatic_control.py using "-a Basic" arguments, the vehicle doesn't go to specified spawn point, it just stops at random location. It doesn't even respect traffic lights as it stated in BasicAgent class.

hongyuli, 2019-03-13:
Same here. 

>  the vehicle doesn't go to specified spawn point, it just stops at random location. It doesn't even respect traffic lights as it stated in BasicAgent class.

dianchen96, 2019-04-05:
I think the issue is related to #1491 , where the global planner gives the wrong high-level command at intersections due to the wrong map topology, making the local planner sample the wrong waypoint. Not 100% sure though

sheepwxy, 2019-04-23:
Does anybody know how to control the car from A to B then to C,I try to do like this 
 ` try:
        client = carla.Client(args.host, args.port)
        client.set_timeout(4.0)

        display = pygame.display.set_mode(
            (args.width, args.height),
            pygame.HWSURFACE | pygame.DOUBLEBUF)

        hud = HUD(args.width, args.height)
        world = World(client.get_world(), hud, args.filter)
        controller = KeyboardControl(world, False)

        if args.agent == "Roaming":
            agent = RoamingAgent(world.player,target_speed=30)
        else:
            agent = BasicAgent(world.player,target_speed=30)    
	    agent.set_destination([100,20,4])
	    agent.set_destination([40,20,4])`
but it seems the car may go wrong if  when I add two set_destination commands at a time.
Can someone help me?

Derekabc, 2019-05-06:
same question? do you solve the problem?

Sakrust, 2019-05-07:
If you use two set_destination commands of course your first target will be overwritten and the car will be rerouted. So you need to test if your vehicle has reached your first target point, this could be achieved by generating a waypoint for both target and vehicle location and comparing their road_ids and distances.

fty0724, 2019-05-08:
@Sakrust Can you show the detail codes how to realize two set_destination commands?

Sakrust, 2019-05-08:
I have no code for this purpose, but you should do something like this:
`agent.set_destination(<location_b>)`

then test every step:
```
waypoint = map.get_waypoint(vehicle.get_location())
target = map.get_waypoint(<location_b>)
if waypoint.transform.location.distance(location_b) < YOUR_VALUE and waypoint.road_id == target.road_id:
agent.set_destination(<location_c>)
```

fty0724, 2019-05-08:
OK,I will have a try, thank you very much

sheepwxy, 2019-05-10:
@Sakrust     I try your code but the Carla window has no response keeping black, can you check what's wrong with this code? Thank you!
`    try:
        client = carla.Client(args.host, args.port)
        client.set_timeout(4.0)

        display = pygame.display.set_mode(
            (args.width, args.height),
            pygame.HWSURFACE | pygame.DOUBLEBUF)

        hud = HUD(args.width, args.height)
        world = World(client.get_world(), hud, args.filter)
        controller = KeyboardControl(world, False)
        if args.agent == "Roaming":
            agent = RoamingAgent(world.player,target_speed=20)
        else:
            agent = BasicAgent(world.player,target_speed=20)
	    agent.set_destination([100,20,4])
	    wpoint = world.map.get_waypoint(world.player.get_location())
	    target = world.map.get_waypoint(carla.Location(x=100,y=20,z=4))
	    if wpoint.transform.location.distance(carla.Location(x=100,y=20,z=4)) < 5.0 and wpoint.road_id == target.road_id:
		agent.set_destination(200,40,4)`

closed 2019-07-16 by stale[bot]