carla-simulator/carla #1339: How to get the 3d corner coordinates and image coordinates of the created model?

backer777, 2019-03-05:
How to get the vertex coordinates and image coordinates of the created model
[image]
But now I want to get the corner data and name of the model (red points in the figure below) ,  how can I get, do you have any hints? Thank you.
[image]

Unlingius, 2019-03-08:
at least you can run make launch and put some test objects to points of interest and look what coordinates it will have. crude, but works i think.

backer777, 2019-03-29:
@Unlingius 
So cool! I got it, Thanks!!

[image]

closed 2019-03-29 by backer777