carla-simulator/carla #1340: Vehicle start steps

Johnnyswong, 2019-03-05:
@nsubiron 
Hi, I have used an rl algorithm to run the Carla simulator. But I have figured out a small trick in your simulator.

The vehicle, which I use is Ford Mustang, needs some speeding steps to move forward.
For example, I need to set the throttle to 1 in the first 100 steps to make the vehicle move. The car won't move if I set the throttle to 1 in less 100 steps. Sometimes 100 steps can't work and you need 130 steps to get it move.

Is there any parameters related to this phenomenon.

Sakrust, 2019-03-05:
Are you sure this is step related? 

Try using a fixed-step solver and a fixed step size. Another thing to mention, right after spawning your vehicle it may take some time to move it forward, especially if you don't spawn it directly on ground level.

Johnnyswong, 2019-03-05:
> Are you sure this is step related?
> 
> Try using a fixed-step solver and a fixed step size. Another thing to mention, right after spawning your vehicle it may take some time to move it forward, especially if you don't spawn it directly on ground level.

Thank you for your reply. So is it a better choice to wait for some seconds before the client send the control signal to the server?

Johnnyswong, 2019-03-05:
BTW, can I move the car to another place with the set_transform in Actor class. The destroy method may sometimes cause the timeout error.

closed 2019-05-11 by stale[bot]