carla-simulator/carla #1341: Carla 0.9.4 Compiled version Town 6 python headers missing.

aroongta, 2019-03-05:
I was trying to run the compile version of Carla 0.9.4, the one with just Town 6; but cannot find a .sh file or python headers in the zip.

nsubiron, 2019-03-06:
The Town06 package contains only the assets (3D models, materials, etc.) and map of Town06, you can add it to your own copy of the simulator. It's a DLC talking in video-game terms. Download the `TownXX.tar.gz`, place it in `ExportedMaps` folder (no need to decompress!), then run the `ImportMaps.sh` script. See [How to import a map](https://carla.readthedocs.io/en/latest/export_import_dist/).

This is the first town that we release with this new method. The idea is to have the towns as separate packages, so you download only what you are going to use (otherwise the main executable is getting too big). Plus, with this, people can now share their custom maps without having to recompile and redistribute the simulator.

Sorry we haven't documented this feature enough!

Manojbhat09, 2019-03-06:
But is it possible to use this asset map in lower versions of Carla?
0.9.1 specifically?
Does the UE version effect it?

aroongta, 2019-03-06:
I see.
Thank you @nsubiron 
Also, is there any way to package a specific map as a standalone simulation server. (Using the package makefile or the packaging function in UE editor). I didn't find anything related in the documentation.

nsubiron, 2019-03-07:
@Manojbhat09 In older versions of Carla you need to recompile the simulator to add a new map. Also using a map from a different version may not work or give errors, it depends if the assets in the map depend on our code.

@aroongta You can set the maps that will be included when doing a `make package` in [DefaultGame.ini](https://github.com/carla-simulator/carla/blob/2e18abb576ff53947c9b7299c3e1c71b5fdbaac7/Unreal/CarlaUE4/Config/DefaultGame.ini#L52-L56).

closed 2019-05-13 by stale[bot]