carla-simulator/carla #1344: Ways to draw the map (and actors)

silgon, 2019-03-06:
I know in `no_rendering_mode.py` you have a way to draw the map in `pygame`. I would like to ask the following questions concerning the drawing of the scene.

- Is there any way to do the information of `no_rendering_mode.py` in a more standard way? within the file you use things such a `carla_map.get_topology()` to get information about the map, shouldn't be with the opendrive standard?
- With `world.get_map().to_opendrive()` I can get the map in the standard format of opendrive, is there a nice way to plot this info? (I cannot find any)
- `carla-ros-bridge` seems to use opendrive in the topic `/carla/map` which is type `string` and not map, therefor it cannot be used in rviz (somehow related to the previous bullet point)
- Is there any work on passing this information in a way such as [`ChauffeurNet`](https://arxiv.org/pdf/1812.03079.pdf) for learning purposes.

closed 2019-05-22 by stale[bot]

MayDGT, 2023-11-24:
> Hi @silgon,
> 
> * There has been some work on exporting all the useful info in `no_rendering_mode.py` as a Python dictionary (road layout, actor transforms, signals, etc); @manishthani and @felipecode, any updates?
> * Functions like `get_topology`, `get_waypoint`, etc. are part of our higher level API to navigate OpenDrive files. They are built exclusively on top of the OpenDrive file provided with the map. The information is exactly the same, we just convert it to something easier to digest for the common use cases (get closest waypoint, get next waypoints, get waypoint on the left, ...) so you don't need to parse the OpenDrive file yourself.
> * `world.get_map().to_opendrive()` dumps the `*.xodr` file as string (also `save_to_disk`), you can use any OpenDrive visualizer to view the road, e.g. [OpenDRIVE Viewer](http://www.opendrive.org/download.html). Alternatively, for a quick visualization with our API you can plot the waypoint locations, to generate waypoints everywhere use `generate_waypoints(distance)`.

Hi, when I use an OpenDRIVE file as the map for Carla, the world.get_map().get_topology() will be empty. Do you know the reason?

MayDGT, 2023-11-24:
> Hi @silgon,
>
> * There has been some work on exporting all the useful info in `no_rendering_mode.py` as a Python dictionary (road layout, actor transforms, signals, etc); @manishthani and @felipecode, [...]

Hi, @nsubiron ,
when I use an OpenDRIVE file as the map for Carla, the **world.get_map().get_topology()** in global planner will be empty. Do you know the reason?