carla-simulator/carla #1345: how to run CARLA0.9.3 with Windows10

xaviera86, 2019-03-07:


xaviera86, 2019-03-07:
I try to run Carla0.9.3 with windows10.
this is my step to run it.
c:\Carla\CARLA_0.9.3_windows>CarlaUE4.exe

but when I excute the client side, I got these errors like below.
=============================================================
c:\Carla\CARLA_0.9.3_windows>python manual_control.py
Traceback (most recent call last):
  File "manual_control.py", line 91, in <module>
    import carla
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\__init__.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
  File "C:\Users\NSW00_907321\AppData\Local\Programs\Python\Python37\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The specified module could not be found. 
=================================================================
is here someone can help me?

marcgpuig, 2019-03-07:
Hi @xaviera86 
Can you try my branch `marcgpuig/zlib_static_link`?

closed 2019-03-07 by nsubiron

jmarrr, 2019-03-31:
Hi, I also encounter this problem on 0.9.4, with windows 10

RoadToML, 2019-04-13:
@marcgpuig  i am encountering this issue too.

downloaded 0.9.4 windows build. 
ran easy_install for the carla-0.9.4-py3.7-win-amd64.egg
tried import carla in a python shell - and get 
```
    from .libcarla import *
ImportError: DLL load failed: The specified module could not be found.
```

please help

MaxHBor, 2019-04-19:
Confirm the issue. Same problem on the same system.

DiegoOGV, 2019-05-02:
The same problem for me. Windows 10 and Carla 0.9.4

DiegoOGV, 2019-05-02:
Problem fixed. I've just installed anaconda on my PC and build the environment with.

marcgpuig, 2019-05-20:
As a workaround, you can try to copy an x64 `zlib.dll` where you have your `.py` files.
If you compiled the project yourself, you will find a `zlib.dll` in `Build/zlib-install/bin`.

MaxHBor, 2019-05-20:
Yes. It works via Anaconda, but according to docs it suppose to work without it, right?

rahulkrishnan20, 2020-09-30:
> As a workaround, you can try to copy an x64 `zlib.dll` where you have your `.py` files.
> If you compiled the project yourself, you will find a `zlib.dll` in `Build/zlib-install/bin`.

I'm using the precompiled windows build version 0.9.10. I'm running into the same issue 
  File "spawn_npc.py", line 24, in <module>
    import carla
  File "C:\Users\Rahul\Documents\WindowsNoEditor\PythonAPI\carla\__init__.py", line 8, in <module>
    from .libcarla import *
  File "C:\Users\Rahul\Documents\WindowsNoEditor\PythonAPI\carla\libcarla.py", line 7, in <module>
    __bootstrap__()
  File "C:\Users\Rahul\Documents\WindowsNoEditor\PythonAPI\carla\libcarla.py", line 6, in __bootstrap__
    imp.load_dynamic(__name__,__file__)
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1776.0_x64__qbz5n2kfra8p0\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed while importing libcarla: The specified module could not be found.

@marcgpuig 
I found the zlib.dll file but I'm not sure where exactly I'm supposed to copy the file into. Could you clarify the what you mean by "where your .py files are" . 

Thanks!

PR #1350 (2019-03-07): Zlib correctly statically linked
#### Description
Some people had issues with the following error while running the client:
```
ImportError: DLL load failed: The specified module could not be found.
```
This is an issue with `zlib` that was not correctly statically linked.
A workaround was putting a x64 `zlib.dll` on `carla/PythonAPI` folder.
This is not necessary anymore because `zlib` is correctly statically linked now.

Fixes #1345

#### Where has this been tested?

  * **Platform(s):** Win10
  * **Python version(s):** 3.x
  * **Unreal Engine version(s):** -

---
This change is [[image]](ht [... cut]