carla-simulator/carla #1346: how can I set the starting position on Carla0.9.3?

xaviera86, 2019-03-07:


dixantmittal, 2019-03-07:
Hi @xaviera86, 
If you mean actor's starting position, then you can either set it while spawning the actor or by using `actor.set_location(Location())` after you have spawned it.
Otherwise, please provide some more details about your issue.

-Dixant

xaviera86, 2019-03-08:
Hi Dixant
I only use CARLA0.9.3 with Compiled version(windows).
so could you tell me how to set the starting position by editing the manual_control.py file?
and how can I set the player?
e.g. Vehicle: Tsla Model3 at some position starting

thanks
xaviera86

dixantmittal, 2019-03-08:
Hi, @xaviera86, 
Thanks for elaborating on your issue. 
`self.map.get_spawn_points()` this gives you a list of pre-defined spawn points for a map. Parameters for `try_spawn_actor(blueprint, transform)` are a `carla.Blueprint()` and a `carla.Transform()` object. So, you can define your own spawn point by creating a new `carla.Transform()` object. 

For your reference:
```
blueprint = blueprint_lib.find('vehicle.tesla.model3')
spawn_point = carla.Transform(carla.Location(x=1, y=2, z=3), carla.Rotation())
self.player = world.try_spawn_actor(blueprint, spawn_point)
```

Note: The spawn point should not collide with other object or building, else spawning will fail.

-Dixant

lijecaru, 2019-03-21:
Hi @xaviera86 @dixantmittal I have found that it is easier to "know" the spawn points if you build Carla from source. That way, when you are in the Unreal Engine Project you can actually see what are the coordinates of the spawn points and can use them to spawn your vehicle in a predefined location. 

If you prefer to stay with the compiled version of the simulator, after you launch the server, you can connect i.e. `manual_control.py` where you can _print_ all the possible spawn points for that particular town, after you use `spawn_points = self.map.get_spawn_points()`. Then you can choose one point in particular, i.e. `spawn_point = spawn_points[4]`, which would define your spawn point as the fifth element in your `spawn_points` array.

dixantmittal, 2019-03-21:
Hi @lijecaru, you can definitely use any spawn point from the list given by `map.get_spawn_points()`. Or, if you want to define your own custom spawn point, you can do that by `spawn_point = carla.Transform(carla.Location(x=1, y=2, z=3), carla.Rotation())`.

closed 2019-05-27 by stale[bot]